rauc/rauc #14: Userinterface is not intuitive

mgrzeschik, 2015-12-09:
When creating a bundle, you have to point to an folder where the
manifest.raucm is available. The filename seems to be fixed. It can
not be changed. You have to suffix the generated bundle with raucb,
otherwise the installer routine will not recognize it as a bundle.

gavinschenk, 2016-06-20:
I used ipkg/opkg in my last PTXDIST-project and build an update process around ipkg-Repositories. As in your solution we bundled multiple ipk-files to a squashfs and provided one file as update. After having some problems with blacklisting/whitelisting some packages the system runs good. 

What do you think about combining ipgk/opkg and RAUC in future -> opkg-handler for RAUC ...

ejoerns, 2016-06-20:
Hi, please open a new Issue/request when discussing topics not related to this issue. Thanks!

ejoerns, 2017-01-31:
Please check #95 and give a short feedback if you find that helpful and sufficient or let us know where we could improve RAUC further concerning this topic (documentation / tooling / warnings).

ejoerns, 2017-06-18:
Closed due to inactivity. Probably fixed.

closed 2017-06-18 by ejoerns