rauc/rauc #18: shorthelp

mgrzeschik, 2016-01-15:
should not contain optional parameters.
Otherwise --key is not optional if generating raucb images

ejoerns, 2017-06-18:
I guess the behavior you described is fixed in current versions of RAUC or could you point out where this still occurs?

Current Output of `rauc -h` is

```
Usage:
  rauc [OPTION...] <COMMAND>

Options:
  -c, --conf=FILENAME               config file
  --cert=PEMFILE                    cert file
  --key=PEMFILE                     key file
  --mount=PATH                      mount prefix
  --override-boot-slot=SLOTNAME     override auto-detection of booted slot
  --handler-args=ARGS               extra handler arguments
  -d, --debug                       enable debug output
  --version                         display version
  -h, --help                        

List of rauc commands:
  bundle	Create a bundle
  checksum	Update a manifest with checksums (and optionally sign it)
  install	Install a bundle
  info		Show file information
  status	Show status
```

ejoerns, 2017-08-08:
Closed due to inactivity and as we assume this issue is already fixed.

closed 2017-08-08 by ejoerns