rauc/rauc #73: Fails to compile on a raspberry pi

blank-frank, 2016-09-14:
./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make supports nested variables... (cached) yes
checking whether to enable debugging... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... 64
checking for a sed that does not truncate output... /bin/sed
checking whether C compiler accepts -Werror=unknown-warning-option... no
checking whether C compiler accepts -fno-strict-aliasing... yes
checking whether C compiler accepts -Wall... yes
checking whether C compiler accepts -Wextra... yes
checking whether C compiler accepts -Wundef... yes
checking whether C compiler accepts -Wnested-externs... yes
checking whether C compiler accepts -Wwrite-strings... yes
checking whether C compiler accepts -Wpointer-arith... yes
checking whether C compiler accepts -Wmissing-declarations... yes
checking whether C compiler accepts -Wmissing-prototypes... yes
checking whether C compiler accepts -Wstrict-prototypes... yes
checking whether C compiler accepts -Wredundant-decls... yes
checking whether C compiler accepts -Wno-unused-parameter... yes
checking whether C compiler accepts -Wno-missing-field-initializers... yes
checking whether C compiler accepts -Wdeclaration-after-statement... yes
checking whether C compiler accepts -Wformat=2... yes
checking whether C compiler accepts -Wold-style-definition... yes
checking whether C compiler accepts -Wcast-align... yes
checking whether C compiler accepts -Wformat-nonliteral... yes
checking whether C compiler accepts -Wformat-security... yes
checking whether C compiler accepts -Wsign-compare... yes
checking whether C compiler accepts -Wstrict-aliasing... yes
checking whether C compiler accepts -Wshadow... yes
checking whether C compiler accepts -Winline... yes
checking whether C compiler accepts -Wpacked... yes
checking whether C compiler accepts -Wmissing-format-attribute... yes
checking whether C compiler accepts -Wmissing-noreturn... yes
[... 78 lines cut ...]
checking whether compiling and linking against OpenSSL works... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating test/services/de.pengutronix.rauc.service
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands

make
make --no-print-directory all-am
  CC       src/librauc_la-bundle.lo
src/bundle.c: In function 'check_bundle':
src/bundle.c:301:5: error: format '%u' expects argument of type 'unsigned int', but argument 5 has type 'guint64' [-Werror=format=]
     "signature size sanity check failed: %"G_GSIZE_FORMAT" exceeds bundle size", sigsize);
     ^
src/bundle.c:308:5: error: format '%u' expects argument of type 'unsigned int', but argument 5 has type 'guint64' [-Werror=format=]
     "signature size sanity check failed: %"G_GSIZE_FORMAT" exceeds 64KiB", sigsize);
     ^
cc1: all warnings being treated as errors
Makefile:1136: recipe for target 'src/librauc_la-bundle.lo' failed
make[1]: *** [src/librauc_la-bundle.lo] Error 1
Makefile:813: recipe for target 'all' failed
make: *** [all] Error 2

ejoerns, 2016-09-15:
Thanks for reporting this issue! :+1:  This is obviously a bug in the printf format specifiers. Prepared #74 which should fix that

jluebbe, 2016-09-17:
I've merged #74, so this should be fixed. @blank-frank, could you verify the fix?

blank-frank, 2016-09-19:
Compiles without errors on RPi, but t will be a little while before I
can actually test-drive it.

Thanks for the quick fix!

jluebbe, 2016-09-19:
Thanks for the confirmation.

closed 2016-09-19 by jluebbe