rauc/rauc #85: rauc install fails with busybox tar: invalid tar magic

gavinschenk, 2016-12-02:
Hi,

I had problems getting rauc work with busybox tar. Using nativ tar rauc just works fine!
Maybe not a bug but worth to mention it in documentation?

with default busybox tar settings
Dec 02 09:36:57 CI4000 rauc[1094]: tar: invalid tar magic

with all tar busybox  options enabled
Dec 02 09:50:16 CI4000 rauc[1094]: tar: unknown typeflag: 0x56

Regards
Gavin

ejoerns, 2016-12-02:
You have to enable `FEATURE_TAR_AUTODETECT` for busybox tar as it has to be able to detect the compression method used for compressing the tar.

This we should mention in the documentation, yes.

Regards, Enrico

gavinschenk, 2016-12-12:
A colleague tested it with FEATURE_TAR_AUTODETECT and it works, thanks!

ejoerns, 2017-02-01:
PR #95 got merged which extends documentation to mention that the tar auto detection feature has to be enabled when using Busybox tar.
Do you think this will be sufficient to better guide new users of RAUC?

closed 2017-02-25 by ejoerns