rauc/rauc #86: Rauc needs Kernel support. Mention this in chapter prerequisite?

gavinschenk, 2016-12-02:
Hi,

I failed to get rauc to work because some leak of kernel support. After switching on loopback support and squasfs, rauc works fine.

-# CONFIG_BLK_DEV_LOOP is not set
+CONFIG_BLK_DEV_LOOP=y
+CONFIG_BLK_DEV_LOOP_MIN_COUNT=8
+# CONFIG_BLK_DEV_CRYPTOLOOP is not set

-# CONFIG_SQUASHFS is not set
+CONFIG_SQUASHFS=y
+CONFIG_SQUASHFS_FILE_CACHE=y
+# CONFIG_SQUASHFS_FILE_DIRECT is not set
+CONFIG_SQUASHFS_DECOMP_SINGLE=y
+# CONFIG_SQUASHFS_DECOMP_MULTI is not set
+# CONFIG_SQUASHFS_DECOMP_MULTI_PERCPU is not set
+# CONFIG_SQUASHFS_XATTR is not set
+CONFIG_SQUASHFS_ZLIB=y
+CONFIG_SQUASHFS_LZ4=y
+CONFIG_SQUASHFS_LZO=y
+CONFIG_SQUASHFS_XZ=y
+# CONFIG_SQUASHFS_4K_DEVBLK_SIZE is not set
+# CONFIG_SQUASHFS_EMBEDDED is not set
+CONFIG_SQUASHFS_FRAGMENT_CACHE_SIZE=3

What do you think about a chapter prerequisite in docs?

Regards
Gavin

ejoerns, 2016-12-02:
We already thought about this, of course, but seems as we did not have managed to add this, yet. Thus, thanks for the hint!  I hope we can extend the documentation soon, as there are many important items still worth being mentioned!

ejoerns, 2017-02-01:
PR #95 got merged which extends documentation to mention that SquashFS and loop device support have to be enabled in kernel.
Do you think this will be sufficient to better guide new users of RAUC?

closed 2017-02-25 by ejoerns