rauc/rauc #114: Solving a use-case where the embded system is just a bunch of squashfs files on a normaly readonly ext4 partition

artizirk, 2017-05-08:
So, currently I have a embedded ARM system that has its 4GB of MMC flash storage split into 3 partitions, one persistent storage and two 1GB ext4 read-only system partitons that each contain a uboot boot script, kernel, bunch of different squashfs files and a initramfs that mounts the current system partition and uses overlayfs to mount squashfs file on top of each other to form a working rootfs.

My initial idea was to specify all the files the ext4 system partition in the `manifest.raucm` like so

```
[update]
compatible=Product Awesome
version=2017-05-08

[file.rootfs/uImage]
filename=uImage

[file.rootfs/dtbs/imx6q-parvus-duracor-310.dtb]
filename=dtbs/imx6q-parvus-duracor-310.dtb

[file.rootfs/uInitrd]
filename=uInitrd

[file.rootfs/00-base.squashfs]
filename=00-base.squashfs

[file.rootfs/10-kernel-modules.squashfs]
filename=10-kernel-modules.squashfs

[file.rootfs/30-static-config.squashfs]
filename=30-static-config.squashfs

[file.rootfs/50-app.squashfs]
filename=00-base.squashfs
```

That of course didn't work as currently (if I understood correctly from reading the source code) rauc can only use `[image.<slot>]` targets for updating the system. One option would be to put all the squashfs files to a `system.tar.gz` file and then let the tar_to_ext4_handler`

ejoerns, 2017-05-09:
Hi, thansk for writing down your use case.

Well, I have a some notes about it.

First of all, the `file` type you used is not designed to work with the bundle update case. You already figured that out. It is more or less a remnant from an alternative 'network' installation method that does not use bundles and that we want to replace with a better mechanism in the future. So first of all, using `[file.slotname/filename]` is not an option for you.

If you intend to update files within a partition, you could simply define a path in your rootfs as a slot and then perform 'image' updates on this pseudo-slot. It will look like:

    [slot.config.0]
    device=/path/in/your/rootfs
    ...

and

    [image.config]
    filename=30-static-config.squashfs

BUT, you intend to do this for skipping slots not to update. There are 2 ways this could be done:

1) Place only those images in your update you intend to change. This solution I would *NOT* recommend as for a robust update you always want to define the entire state of your system and not only an untested loose collection of components!!
2) Rely on RAUCs ability to skip updates for slots that are already installed in the exact same version on the target slot. This unfortunately does not work yet with non-writable images, such as a SquashFS as currently the information about the slot status (i.e. its sha1) is only stored in the updated slot itself.

Thus, from my point of view, the `system.tar.gz` option seems to be the most suitable for now. What kind of issues did you encounter with that?

We have plans for different features that will solve parts of your problem (e.g. a central slot metadata store, delta-updates, etc.) but for now, these are not yet implemented.

ejoerns, 2017-08-23:
Any news here? Otherwise I would close this thread within the next days.

closed 2017-09-14 by ejoerns