rauc/rauc #125: `rauc install` fails when `rauc service` is started by dbus

hundeboll, 2017-06-06:
If I run `rauc service -d &` prior to running `rauc install my-bundle.raucb` everything works as expected:

```
root@armada-388-db:~# rauc service -d &
root@armada-388-db:~# Domains: 'rauc'
(rauc:2089): rauc-DEBUG: service start
(rauc:2089): rauc-DEBUG: No mount prefix provided, using /mnt/rauc/ as default
(rauc:2089): rauc-DEBUG: name 'de.pengutronix.rauc' acquired

root@armada-388-db:~# rauc install rauc-bundle-armada-388-db.raucb 
trying to contact rauc service
input bundle: /home/root/rauc-bundle-armada-388-db.raucb
Active slot bootname: A
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: installing
(rauc:2089): rauc-DEBUG: thread started for /home/root/rauc-bundle-armada-388-db.raucb

rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: started
(rauc:2089): rauc-DEBUG: Found booted slot: system.0 on /dev/mmcblk0p1
rauc-Message: Mounting bundle '/home/root/rauc-bundle-armada-388-db.raucb' to '/mnt/rauc/bundle'
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: Checking and mounting bundle...
rauc-Message: Reading bundle: /home/root/rauc-bundle-armada-388-db.raucb
rauc-Message: Verifying bundle... 
Adding to target group: system -> system.1
Target Group:
  system -> system.1
Using default handler
rauc-Message: Marking target slot as non-bootable...
rauc-Message: Checking image type for slot type: ext4
rauc-Message: Image detected as type: *.ext4
rauc-Message: mounting slot /dev/mmcblk0p2
rauc-Message: installing /home/root/rauc-bundle[ 2522.394903] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
-armada-388-db.raucb: Updating slots...
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: Checking slot system.1
rauc-Message: Skipping update for correct image /mnt/rauc/bundle/rauc-image-armada-388-db.ext4
(rauc:2089): rauc-DEBUG: unmounting slot /dev/mmcblk0p2
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: Skipping update for correct image /mnt/rauc/bundle/rauc-image-armada-388-db.ext4
rauc-Message: Marking slots as bootable...
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: Updating slot system.1 done
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: All slots updated
(rauc:2089): rauc-DEBUG: thread finished for /home/root/rauc-bundle-armada-388-db.raucb

rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: finished
rauc-Message: installing `/home/root/rauc-bundle-armada-388-db.raucb` succeeded
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: idle
installing `/home/root/rauc-bundle-armada-388-db.raucb` succeeded
```

If I don't run `rauc service` prior to `rauc install`, the install fails:
```
root@armada-388-db:~# rauc install rauc-bundle-armada-388-db.raucb 
trying to contact rauc service
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: installing
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: idle
installing `/home/root/rauc-bundle-armada-388-db.raucb` failed
```

I tried then modified dbus to call a wrapper when starting rauc, so debug output is logged to a file:
```
root@armada-388-db:~# cat /tmp/rauc.log 
Domains: 'rauc'
(rauc:2115): rauc-DEBUG: service start
(rauc:2115): rauc-DEBUG: No mount prefix provided, using /mnt/rauc/ as default
(rauc:2115): rauc-DEBUG: name 'de.pengutronix.rauc' acquired
input bundle: /home/root/rauc-bundle-armada-388-db.raucb
Active slot bootname: A
(rauc:2115): rauc-DEBUG: thread started for /home/root/rauc-bundle-armada-388-db.raucb

(rauc:2115): rauc-DEBUG: Found booted slot: system.0 on /dev/mmcblk0p1
Adding to target group: system -> system.1
Target Group:
  system -> system.1
Using default handler
```

It just hangs there. If I then kill `rauc service`, the log file is flushed with:
```
(rauc:2115): rauc-DEBUG: thread finished for /home/root/rauc-bundle-armada-388-db.raucb
```

This is on a pretty standard yocto, and it happens on both morty and pyro.

Any suggestions are appreciated.

mnhu, 2017-06-06:
Oh, the current version is `0.1.1`:
```
root@armada-388-db:~# rauc --version
rauc 0.1.1
```

ejoerns, 2017-06-07:
Mh, from the log it seems to stop somewhere around the compatible check. Do you use the standard compatible check or an install-check hook for this?

Maybe its worth trying to pick #126 and see if it provides you any details about what happened at the client side, especially why it assumes to have 'failed'.

mnhu, 2017-06-07:
I'm using the standard check. Also tried to change the compatible string to provoke a mismatch, but it made no difference.

I will try to pick the dbus-error-logging tonight...

mnhu, 2017-06-07:
Testing with #126 does reveal a little more:
```
root@armada-388-db:~# rauc install rauc-bundle-armada-388-db.raucb 
trying to contact rauc service                             
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: installing                                        
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 020% Determining slot states done.                
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 020% Checking bundle                              
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 020% Verifying signature                          
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 040% Verifying signature done.                    
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 040% Checking bundle done.                        
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 050% Verifying manifest checksums                 
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 060% Verifying manifest checksums done.           
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 060% Verifying manifest done.                     
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 060% Determining target install group             
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 080% Determining target install group done.       
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: 100% Installing failed.                           
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: LastError: Handler error: Failed marking slot system.0 non-bootable
rauc-Message: installing /home/root/rauc-bundle-armada-388-db.raucb: idle                                              
installing `/home/root/rauc-bundle-armada-388-db.raucb` failed
```
Still it works when starting `rauc service` prior to running `rauc install` (logs similar to the working log in the first post).

mnhu, 2017-06-07:
Yep, using full path when calling `fw_setenv` makes it work without starting `rauc service` first:

```
git diff        
diff --git a/src/bootchooser.c b/src/bootchooser.c         
index d45ca9b..428c7f0 100644                              
--- a/src/bootchooser.c      
+++ b/src/bootchooser.c      
@@ -11,7 +11,7 @@            
 #define BAREBOX_STATE_ATTEMPS_PRIMARY  3                  
 #define BAREBOX_STATE_DEFAULT_PRIORITY 10                 
 #define BAREBOX_STATE_PRIORITY_PRIMARY 20                 
-#define UBOOT_FWSETENV_NAME "fw_setenv"                   
+#define UBOOT_FWSETENV_NAME "/sbin/fw_setenv"             
                             
 #if 0                       
 static gboolean barebox_state_get_int(const gchar* name, int *value) {
```

ejoerns, 2017-06-07:
Yes, great to see you already worked out the issue with the help of #126, and thanks for testing in different scenarios. Seems as if we should merge it soon.

The issue you revealed is, that when started via D-Bus activation, there is no shell context, or more detailed, a clean environment that does not have any `PATH` set. Thus looking up `fw_setenv` fails. We should consider allowing to use a (configurable) absolute path for these cases.

mnhu, 2017-06-08:
It would be great if you could merge #126 soonish... I noticed that grubenv had an config option to set a path, so maybe a similar thing is needed for uboot (and barebox), or a general path config for all of them.

jluebbe, 2017-06-13:
I'd like to avoid hard-coding the path in rauc, as one tool may be located in /bin on one distribution and /sbin on another. systemd doesn't remove PATH from the environment when activating via D-Bus. If specifying the PATH in D-Bus .service files is not supported, we could have a small wrapper shell script to set it with the same values as systemd.

mnhu, 2017-06-19:
Agreed, hard-coded paths should be avoided.

It doesn't seem like there is a way to set the PATH environment variable in D-Bus service files, so I am looking into installing a wrapper script in `/usr/libexec/rauc-service.sh`.

According to https://github.com/systemd/systemd/blob/master/src/basic/path-util.h#L34 the default systemd-path is `/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin`.

Would something like this work:
```sh
#!/bin/sh
export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
@bindir@/rauc service
```

And then update `/usr/share/dbus-1/system-services/de.pengutronix.rauc.service` to call `/usr/libexec/rauc-service.sh`.

mnhu, 2017-07-19:
This is fixed in https://github.com/rauc/rauc/commit/dcd184d7c89c78c5ed957a02169e836a236d4ece

ejoerns, 2017-08-08:
:+1: Great work done! I guess we can close this one now.

closed 2017-08-08 by ejoerns