rauc/rauc #138: Lack of Barebox Example

jovere, 2017-07-25:
I apologize if this isn't the correct location, but I didn't see anywhere else to post.  If there is a mailing list or discussion forum available, please direct me towards it.

I have been working through using RAUC with Barebox for the last couple days, but it feels like I'm just grasping at straws.  Nowhere in the documentation is clear instructions or an example of how to get RAUC working with Barebox.  The documentation states "If in doubt about choosing the right bootloader, we recommend to use Barebox as it provides a dedicated boot handling framework, called bootchooser.", yet the only documentation about Barebox is compiling it with a couple flags, and putting "barebox" in the system.conf file.  How does it access the "state"?  Where is the "state" stored?  What variables are needed in the "state" so RAUC can do its thing?  Also, the bootchooser documentation seems to show that it has a "state" framework (http://www.barebox.org/doc/latest/user/state.html), but it doesn't show how to *create* one.

A slightly separate issue is that I can't seem to figure out the correct "type" to specify for a ubi partition with the raw kernel/device tree copied into it.  It looks like the table is a "todo" that hasn't been completed yet.  If it helps, I use "ubiupdatevol" to copy the kernel/device tree into their respective partitions.

I feel that if I could just have one example configuration for how the "state" is used with Barebox, and how it's accessed by RAUC, it would go a significant way in helping me move along.

-Jeremy

ejoerns, 2017-07-31:
Hi, you do not need to apologize, this is a good place for starting a discussion. If you prefer email in the future, there is also a mailing list and a few other ways to contact us: http://rauc.io/pages/support.html.

If you have trouble with using RAUC and barebox state there definitely is a lack of documentation somewhere and we know the some points are still not as well documented as they should. We try to put further effort in it, but as you might know, unfortunately, having time for writing documentation is always a crucial point.
Thus, if we know more concrete what is missing already helps us a little in improving the documentation more targeted. Nevertheless, the best way to help improving the documentation is sending pull-requests that improves or extends some paragraphs at least a little.

Now, to focus on your questions:

> How does it access the "state"?

RAUC accesses the state via the utility `barebox-state` from our dt-utils git. This is basically what the documentation already tells you. The information where the state is stored comes from the devicetree. The Barebox devicetree must have a node for the "barebox,state" compatible that describe the states storage location and the variables the state should contain. To let the linux userspace have exactly the same information about the state that barebox has, Barebox fixups the Linux devicetree during booting and inserts the state node and an additional alias, which is also called 'state' by default. Now the barbeox-state tool inspects the devicetree information and reads the barebox,state node referenced by the alias 'state' (default).

For Bootchooser, there is a special set of variables required the state must contain. As a basic example for two redundant boot slots you could take:

    state: state {
            magic = <0x4d433230>;
            compatible = "barebox,state";
            backend-type = "raw";
            backend = <&state_storage>;
            backend-stridesize = <0x40>;
            backend-storage-type = "circular";
            #address-cells = <1>;
            #size-cells = <1>;

            bootstate {
                    #address-cells = <1>;
                    #size-cells = <1>;

                    system0 { /* the node's name here must match the subnode's name in the 'bootstate' node */
                            #address-cells = <1>;
                            #size-cells = <1>;

                            remaining_attempts@0 {
                                    reg = <0x0 0x4>;
                                    type = "uint32";
                                    default = <3>;
                            };
                            priority@4 {
                                    reg = <0x4 0x4>;
                                    type = "uint32";
                                    default = <20>;
                            };
                    };

                    system1 { /* the node's name here must match the subnode's name in the 'bootstate' node */
                            #address-cells = <1>;
                            #size-cells = <1>;

                            remaining_attempts@8 {
                                    reg = <0x8 0x4>;
                                    type = "uint32";
                                    default = <3>;
                            };
                            priority@c {
                                    reg = <0xC 0x4>;
                                    type = "uint32";
                                    default = <20>;
                            };
                    };

                    last_chosen@10 {
                            reg = <0x10 0x4>;
                            type = "uint32";
                     };
             };
     };

You'll also need

    aliases {
            state = &state;
    };

Thate backend then could be an eMMC partition, an EEPROM, a NAND or something different. Barebox state supports a variety of variants here. An example of how to set up state for different storage backends you can find in the Barebox source code at `arch/arm/dts/state-example.dtsi`.

In Barebox you also need to configure the behavior of Bootchooser via some nv variables, but this should be described a bit more verbose in http://barebox.org/doc/latest/user/bootchooser.html.

The important link between the different instances are the names of the bootchooser targets.

1) You define them in Barebox nv via

        nv bootchooser.targets="system0 system1"

2) You connect them to a Barebox boot target (automagically one or custom script)

        nv bootchooser.system0.boot=nand0.ubi.system0
        nv bootchooser.system1.boot=nand0.ubi.system1

3) You have to create a node in the bootstate state node for each of them (as shown above).

        state {
          bootstate {
            system0 {
            ...
            };
            system1 {
            ...
            };
          };
        };

4) You also have to set their names in the `bootname` property of RAUCs system.conf in order to let barebox-state access the correct variables.

```
[slot.rootfs.0]
device=/dev/ubi0_0
type=ubivol
bootname=system0

[slot.rootfs.1]
device=/dev/ubi0_1
type=ubivol
bootname=system1

```

Then, if you boot you system with barebox bootchooser, i.e. 

    boot bootchooser

Barebox will tell RAUC the currently booted slot by setting `bootchooser.active=systemX` in the kernel command line and RAUC and Barebox will interact for installing images, manipulating attempts counters, etc.

Hope that helps at least a little?

jovere, 2017-08-01:
That actually clears up quite a bit.  However, I noticed that I needed to manually create the `alias`, and therefore an empty `state` section in my kernel device tree, as the alias wasn't being passed from barebox.

There were also missing parts to the configuration of Barebox.  I found that I needed to include *all* of the following items in the kconfig for it to work properly.
```
CONFIG_STATE=y
CONFIG_BOOTCHOOSER=y
CONFIG_CMD_STATE=y
CONFIG_CMD_BOOTCHOOSER=y
CONFIG_STATE_DRV=y
```

I am now able to implement an update of the root file system.

ejoerns, 2017-08-02:
Good to hear that you managed to make it work basically!

Which Barebox version do you use? This should be fixed in Barebox-commit 70ea799f6b6ddcd92b94420a2d352a73f0f9ff08 ('state: Create alias in of_state_fixup()') which made its way into v2017.06.

> CONFIG_STATE=y
> CONFIG_BOOTCHOOSER=y
> CONFIG_CMD_STATE=y
> CONFIG_CMD_BOOTCHOOSER=y
> CONFIG_STATE_DRV=y

Yes, this is the full set. As bootchooser and state are more or less independent from each other, they will not be automatically selected. But it is the most common case to use them all together.

MDM23, 2017-08-16:
Sorry for interrupting here, but I'm also a bit stuck with implementing rauc using barebox. I know this is not exactly the right location, as I just got some difficulties getting barebox-state to work, but I hope that you could give me a small hint. I could not find much documentation about this topic  :confused:

I got everything set up and now I need to enable barebox-state using a NAND partition. I already patched the device tree within the barebox source code. These are the changes that I applied:

```dts
/ {
  aliases {
    state = &state;
  };

  state: state {
    magic = <0x27031977>;
    compatible = "barebox,state";
    backend-type = "raw";
    backend = <&state_part>;
    backend-stridesize = <0x40>;
    backend-storage-type = "circular";

    bootstate {
      system0 {
        remaining_attempts@0 {
          reg = <0x0 0x4>;
          type = "uint32";
          default = <3>;
        };
        priority@4 {
          reg = <0x4 0x4>;
          type = "uint32";
          default = <20>;
        };
      };

      system1 {
        remaining_attempts@8 {
          reg = <0x8 0x4>;
          type = "uint32";
          default = <3>;
        };
        priority@c {
          reg = <0xC 0x4>;
          type = "uint32";
          default = <20>;
        };
      };

      last_chosen@10 {
        reg = <0x10 0x4>;
        type = "uint32";
      };
    };
  };
};
```

Furthermore, I added the partition that I want to use:

```dts
&gpmc {

  /* ... */

  nand: nand@0,0 {

    /* ... */

    state_part: partition@1C0000 {
      label = "barebox_state";
      reg = <0x1C0000 0x40000>;
    };

    /* ... */

  };
};
```

Somehow, it seems to work on barebox side. After booting, the remaining_attempts variable is decremented and stored. I get these error messages though:

```
state: state failed to parse path to backend: No such device
state: Failed to initialize desired amount of buckets, only 2 of 3 succeeded
state: New state registered 'state'
state: Using bucket 0@0x00000000
```

When running the barebox-state command in Linux, it displays:

```
of_get_devicepath: cannot find device from node /ocp/gpmc@50000000/nand@0,0/partitions
Cannot find backend path in /state
```

Do you know what could possibly be the issue?

Thank you very much in advance,
Peter

ejoerns, 2017-08-16:
FYI, in #144 we extended the RAUC documentation regarding a huge set of aspects. One of it was the Barebox integration. Please check 

  https://github.com/rauc/rauc/blob/master/docs/integration.rst#barebox

We also heavily extended the Barebox state and bootchooser documentation, the patch is already on the mailing list.

  http://lists.infradead.org/pipermail/barebox/2017-August/030956.html

ejoerns, 2017-08-16:
@MDM23 if you are using NAND, you may omit the 'stridesize' setting as it will have no effect. State will use the NAND's page size as the stride size.

To let all 3 redundant state copies fit into the state partition, the state partition size must be at least 3 times the size of the NAND's eraseblock size. Due to possible bad blocks it is recommended to reserve at least 6 times the eraseblock size.

For allowing the userspace to detect the NAND partition, make sure to have the same NAND partitions available in Linux by removing NAND partition form linux DTB and letting barebox fixup the partitions to the Linux DTB.

Does that help?

MDM23, 2017-08-16:
@ejoerns thank you for your fast help! The warning within barebox disappeared after setting a bigger partition size.

I checked the device tree within Linux and could not spot any issue, so I started debugging dt-utils on my target and found a working configuration. I looked at the function `int of_get_devicepath( /* ... */ )` in `libdt.c` and dumped the contents of `partition_node->parent` right before this part (line 2308 in current master):

```c
/*
* Ok, the partition node has no udev_device. Try parent node.
*/

node = partition_node->parent;
```

The output was something like:

```
partitions {
        compatible = "fixed-partitions";
        #size-cells = <0x1>;
        #addres-cells = <0x1>;
        partition@0 {
        // ...
```

I thought this should point to the nand node, not its partitions, so I gave it a try and changed the code to:

```
node = partition_node->parent->parent;
```

That did the trick and barebox-state worked fine! Did I miss something or made a configuration error?

BastianStender, 2017-08-16:
Hi,

are you using the latest dt-utils version? This should be solved by https://git.pengutronix.de/cgit/tools/dt-utils/commit/?id=ea1d08e26b73a4855165ed56043fa439ff9948cb

```
	/*
	 * Respect flash "partitions" subnode. Use parent of parent in this
	 * case.
	 */
	if (!strcmp(node->name, "partitions"))
		node = node->parent;
```

MDM23, 2017-08-17:
@BastianStender seems like our board vendor used an old version of dt-utils. After replacing it with the newest one, everything works fine!

Thank you very much for your help!  :+1:

ejoerns, 2017-08-18:
Any issues left here? Did someone take a look at our new and shiny documentation? ;) 

  http://rauc.readthedocs.io/en/latest/integration.html#interfacing-with-the-bootloader

jovere, 2017-08-18:
I'd say it looks good.  The only thing I'd add is a link to the barebox state devicetree documentation, which for some reason, barebox doesn't have at all in their bootchooser documentation.  It would have saved me a lot of time searching if there was an actual link:
http://barebox.org/doc/latest/devicetree/bindings/barebox/barebox,state.html

ejoerns, 2017-08-20:
Thanks, good hint! I tried to address this in #149

ejoerns, 2017-08-23:
#149 got merged, thus I will close this thread for now.

leiflm, 2017-11-14:
I think @BastianStender's [comment](https://github.com/rauc/rauc/issues/138#issuecomment-322789733) is **very** important here. The documentation should mention that you might need a certain revision of dt-utils or - at least - a certain version.

ejoerns, 2017-11-14:
Seems to me GitHub is a bit secretive in highlighting changes to closed Issues, thus I will reopen it untill the topic is resolved again.
Did you run into the same issues by using an outdated version of dt-utils?

ejoerns, 2017-11-14:
Now addressed by #187

leiflm, 2017-11-14:
> Did you run into the same issues by using an outdated version of dt-utils?

Yes, indeed, I did. Thanks for picking this up :-)

ejoerns, 2017-11-15:
#187 got merged, thus I will close this again.

closed 2017-11-15 by ejoerns

ShoaibAhmad123, 2023-01-25:
> Sorry for interrupting here, but I'm also a bit stuck with implementing rauc using barebox. I know this is not exactly the right location, as I just got some difficulties getting barebox-state to work, but I hope that you could give me a small hint. I could not find much documentation about this topic confused
> 
> I got everything set up and now I need to enable barebox-state using a NAND partition. I already patched the device tree within the barebox source code. These are the changes that I applied:
> 
> ```
> / {
>   aliases {
>     state = &state;
>   };
> 
>   state: state {
>     magic = <0x27031977>;
>     compatible = "barebox,state";
>     backend-type = "raw";
>     backend = <&state_part>;
>     backend-stridesize = <0x40>;
>     backend-storage-type = "circular";
> 
>     bootstate {
>       system0 {
>         remaining_attempts@0 {
>           reg = <0x0 0x4>;
>           type = "uint32";
>           default = <3>;
>         };
>         priority@4 {
>           reg = <0x4 0x4>;
>           type = "uint32";
>           default = <20>;
>         };
>       };
> 
>       system1 {
>         remaining_attempts@8 {
>           reg = <0x8 0x4>;
>           type = "uint32";
>           default = <3>;
>         };
>         priority@c {
>           reg = <0xC 0x4>;
>           type = "uint32";
>           default = <20>;
>         };
>       };
> 
>       last_chosen@10 {
>         reg = <0x10 0x4>;
>         type = "uint32";
>       };
>     };
>   };
> };
> ```
> 
> Furthermore, I added the partition that I want to use:
> 
> ```
> &gpmc {
> 
>   /* ... */
> 
>   nand: nand@0,0 {
> 
>     /* ... */
> 
>     state_part: partition@1C0000 {
>       label = "barebox_state";
>       reg = <0x1C0000 0x40000>;
>     };
> 
>     /* ... */
> 
>   };
> };
> ```
> 
> Somehow, it seems to work on barebox side. After booting, the remaining_attempts variable is decremented and stored. I get these error messages though:
> 
> ```
> state: state failed to parse path to backend: No such device
> state: Failed to initialize desired amount of buckets, only 2 of 3 succeeded
> state: New state registered 'state'
> state: Using bucket 0@0x00000000
> ```
> 
> When running the barebox-state command in Linux, it displays:
> 
> ```
> of_get_devicepath: cannot find device from node /ocp/gpmc@50000000/nand@0,0/partitions
> Cannot find backend path in /state
> ```
> 
> Do you know what could possibly be the issue?
> 
> Thank you very much in advance, Peter

hi, sorry ft inconvience, but i stuck on same issue, when i run barebox-state in linux i get an error 
" Neither /aliases/state nor /state found" 
i didn't understand what is the issue.
Thanks in advance...

ejoerns, 2023-01-25:
hank you very much in advance, Peter
> 
> hi, sorry for inconvenience, but i stuck on same issue, when i run barebox-state in linux i get an error " Neither /aliases/state nor /state found" i didn't understand what is the issue. Thanks in advance...

Which barebox version are you using?

If I remember correctly, one needed to have an empty state node and alias in linux devicetree for old barebox versions for the fixup to apply correctly

ShoaibAhmad123, 2023-01-25:
> 

thanks for the reply. My barebox version is "2018.11.0-phy3-r7.0". according to barebox definition 
```
Since the state variable set is intended to be shared between the bootloader and the kernel, the view to the state variable set must be the same in both worlds.

This can be achieved by defining all state variable set related definitions inside the barebox’s devicetree only. It’s not required to keep and maintain the same information inside the Linux kernel’s devicetree again.
```