rauc/rauc #165: rauc status mark-good/bad permission with permission denied

brainsprinter, 2017-09-07:
rauc status mark-good/bad command fails with "Operation not permitted" error. Installing update bundle fails too. I can however set u-boot environment variable manually using fw_setenv command. I am using meta-rauc with yocto.

`Sep 07 14:03:26 colibri-imx6 rauc[1045]: rauc-Message: installing `/media/sda1/update-bundle-colibri-imx6-20170907135421.raucb` failed: 1
Sep 07 14:03:26 colibri-imx6 rauc[1045]: rauc-Message: installing /media/sda1/update-bundle-colibri-imx6-20170907135421.raucb: finished
Sep 07 14:03:26 colibri-imx6 rauc[1045]: rauc-Message: installing /media/sda1/update-bundle-colibri-imx6-20170907135421.raucb: Handler error: Failed marking slot rootfs.1 non-bootable
Sep 07 14:03:26 colibri-imx6 rauc[1045]: (rauc:1045): rauc-WARNING **: Handler error: Failed marking slot rootfs.1 non-bootable
Sep 07 14:03:26 colibri-imx6 rauc[1045]: (rauc:1045): rauc-WARNING **: failed marking as bad
Sep 07 14:03:26 colibri-imx6 rauc[1045]: (rauc:1045): rauc-WARNING **: fw_setenv failed: Child process exited with code 1
Sep 07 14:03:26 colibri-imx6 rauc[1045]: Error: can't write fw_env to flash
Sep 07 14:03:26 colibri-imx6 rauc[1045]: Write error on /dev/mmcblk0boot0: Operation not permitted
Sep 07 14:03:26 colibri-imx6 rauc[1045]: rauc-Message: Marking target slot as non-bootable...
Sep 07 14:03:25 colibri-imx6 rauc[1045]: Using default handler
Sep 07 14:03:25 colibri-imx6 rauc[1045]:   rootfs -> rootfs.1
Sep 07 14:03:25 colibri-imx6 rauc[1045]: Target Group:
Sep 07 14:03:25 colibri-imx6 rauc[1045]: Adding to target group: rootfs -> rootfs.1
Sep 07 14:03:03 colibri-imx6 rauc[1045]: rauc-Message: Verifying bundle...
Sep 07 14:03:03 colibri-imx6 rauc[1045]: rauc-Message: Reading bundle: /media/sda1/update-bundle-colibri-imx6-20170907135421.raucb
Sep 07 14:03:03 colibri-imx6 rauc[1045]: rauc-Message: installing /media/sda1/update-bundle-colibri-imx6-20170907135421.raucb: Checking and mounting bundle...
Sep 07 14:03:03 colibri-imx6 rauc[1045]: rauc-Message: Mounting bundle '/media/sda1/update-bundle-colibri-imx6-20170907135421.raucb' to '/run/rauc/bundle'
Sep 07 14:03:03 colibri-imx6 rauc[1045]: rauc-Message: installing /media/sda1/update-bundle-colibri-imx6-20170907135421.raucb: started
Sep 07 14:03:03 colibri-imx6 rauc[1045]: Active slot bootname: A
Sep 07 14:03:03 colibri-imx6 rauc[1045]: input bundle: /media/sda1/update-bundle-colibri-imx6-20170907135421.raucb
`

ejoerns, 2017-09-14:
Did you verify writing to the same partition when trying manually compared to what RAUC attempts to write to?

Is write access enabled on your boot partition?
https://www.kernel.org/doc/Documentation/mmc/mmc-dev-parts.txt

brainsprinter, 2017-09-21:
Thank you for your help. Enabling write permission on boot partitions (two of them) resolved the problem and mark good service and update are working fine now. But I still do not understand why is it possible to do "fw_setenv BOOT_A_LEFT 3" even if the boot partitions are write protected? I took a look at the souce code I saw the rauc is also using the same command for his mark-good service.

ejoerns, 2017-12-04:
Unfortunately, I haven't a clue why it works with manually execution the command on the exact same boot and system.
Maybe for any reasons it uses a different config, is influenced by a different execution environment, etc... 
As long as not having a system here where we can reproduce this, it will also be hard to debug for us.
Thus, if you should find out what the difference is, please let us know.

ejoerns, 2018-12-10:
@brainsprinter is this issue still valid or have you been able to fix this?

brainsprinter, 2018-12-21:
> 
> 
> @brainsprinter is this issue still valid or have you been able to fix this?

@ejoerns yes, I was able to solve this issue. Thx

Psykar, 2020-01-30:
In the interests of helping those who google - in my case my BSP had a function set in `/etc/profile.d/fw_unlock_mmc.sh`

```
# Give fw_setenv mmcblk0boot0 write permissions
fw_setenv() {
    echo 0 > /sys/block/mmcblk0boot0/force_ro
    /sbin/fw_setenv "$@"
    echo 1 > /sys/block/mmcblk0boot0/force_ro
}
```
And of course rauc was executing the binary directly instead of using this function.

ejoerns, 2020-01-30:
I'll reopen this issue as this seems to affect multiple users and obviously either lacks documentation, better error reporting or better handling.

The question that I raised the last time when this was discussed on the mailing list was if it is a good idea to store the environment on the boot partition.

@Psykar Could you describe the setup that you use? Do you update your bootloader using eMMC boot parition switching? Or do you just use boot0 as a fixed location?

From my point of view, the valid approach would be to store the env outside the boot partition and allow the device to update the bootloader by switching between boot0 and boot1.

Psykar, 2020-01-30:
IMO env variables seem fine where they are - and it makes sense to use the kernel ro locking mechanism to prevent accidental changes. 
It might be possible to add get rauc to unset/set ro when it does it's changes, if fw_setenv is configured to use an emmc boot device? I haven't looked at the mailing list though, possibly I'm misunderstanding your suggestion.
My workaround will likely be to wrap any calls to the rauc cli in a unset/set cli tool (similar to the above bash function)

---
In terms of my specific setup (if it's relevant): I'm using a Toradex IMX7, starting from their yocto meta package and adding in rauc-meta.

My current setup (still somewhat WIP)

/dev/mmcblk0boot0  - uBoot & uBoot env variables (toradex's default)
/dev/mmcblk0boot1  - unknown what their default is... I have plans to use this for atomic env updates, but looks like this might not be possible with uBoot (because it's a different partition)
/dev/mmcblk0p1     - a uBoot script (which the env variables point to and execute) - currently still contains my dtb and kernel, but I'll be moving them inside the rootfs soon.
/dev/mmcblk0p2     - root.A
/dev/mmcblk0p3     - root.B
/dev/mmcblk0p4     - data

I've created the bash wrapper for the rauc commands, and that works fine. I'll need to patch the init script if I want the default mark-good to work - but I plan on putting that inside our application rather than immediately after boot, so that won't be a problem.

ejoerns, 2020-01-31:
> /dev/mmcblk0boot0 - uBoot & uBoot env variables (toradex's default)
> /dev/mmcblk0boot1 - unknown what their default is... I have plans to use this for atomic env updates, but looks like this might not be possible with uBoot (because it's a different partition)

That's exactly the point and the reason why I consider the boot partition(s) to be the wrong location for the environment.  When you use them for atomic bootloader updates, you need to switch the boot partitions after having written your bootloader image. If you then had the environment in the boot partition, it would get lost together with its entire state information saved up till then.

Also, switching boot partition is not something the bootloader needs to be set up for, it is the ROM loader of your SoC that decides (baes on the corresponding eMMC's extCSD register value) which partition to boot. This is more or less transparent for the bootloader.

Psykar, 2020-01-31:
So the alternative is to put it on one of the main emmc partitions then I suppose?
The issue with this is just that the default for most BSP's seems to be to use the emmc boot partition, so  more work required to update a BSP to use rauc.
I don't think any of this can be changed in rauc's code base though - only updated docs suggesting to move the uboot env storage to a different partition if the BSP has put it there.

jonaskgandersson, 2020-10-21:
Hi
I have the exact same situation on imx8 based device, BSP default u-boot env on /dev/mmcblk2boot0.

After following this thread I can see that there is reason to change the location of the environment.

In my BSP a shell script like above is supplied to wrap fw_setenv to handle read only partition. I use Yocto to build rauc and added a patch to bootchooser.c to change UBOOT_FWSETENV_NAME to the setenv function supplied from the BSP. I guess an approach could be to enable an option in  system.conf to set custom function to read/write environment?

jluebbe, 2023-05-11:
On platforms which store the environment on eMMC boot partitions (which we'd recommend against, see https://github.com/rauc/rauc/issues/165#issuecomment-580630437), any use of fw_setenv needs to handle the write-protect mechanism via sysfs. Accordingly, I'd argue that the proper fix would be to teach fw_setenv how to boot partitions. It would then be transparent to any existing software using fw_setenv.

Bartel-C8, 2025-02-10:
For others landing here with the same problem: I had the same problem where boot0 and boot1 partitions are RO.
And I am now using another fw_printenv/fw_setenv repo, which automatically locks and unlocks the device to be written to:
https://github.com/sbabic/libubootenv