rauc/rauc #171: signature error for bundle

aapurohit, 2017-09-15:
with the latest version of rauc available on github..

root@arm:~# rauc info /media/usb/testUpdate.raucb 
rauc-Message: Reading bundle: /media/usb/testUpdate.raucb
rauc-Message: Verifying bundle... 
**
rauc:ERROR:src/signature.c:320:cms_get_cert_chain: assertion failed: (sk_X509_value(signers, 0) == sk_X509_value(*verified_chain, 0))
Aborted

I have created certificate and key using :
 openssl req -x509 -newkey rsa:4096 -nodes -keyout demo.key.pem -out demo.cert.pem -subj "/O=rauc Inc./CN=rauc-demo"

I have given the demo.cert.pem as keyring to the system.conf file on target.

P.S. : I have downloaded the older version where I am not facing this problem.

ejoerns, 2017-09-15:
Seems as if the assumption made in the assertion is not valid for all cases, such as self-signed certificates

ejoerns, 2017-09-15:
#172 addresses this issue

ejoerns, 2017-09-15:
#172 got merged, please verify if this also works for you

aapurohit, 2017-09-15:
works for me, thanks

closed 2017-09-15 by aapurohit