rauc/rauc #173: Failed marking slot rootfs non-bootable

aapurohit, 2017-09-15:
root@arm:~# rauc install /media/usb/up2.raucb 
Trying to contact rauc service
rauc-Message: installing /media/usb/up2.raucb: installing
rauc-Message: installing /media/usb/up2.raucb:   0% Installing
rauc-Message: installing /media/usb/up2.raucb:   0% Determining slot states
rauc-Message: installing /media/usb/up2.raucb:  20% Determining slot states done.
rauc-Message: installing /media/usb/up2.raucb:  20% Checking bundle
rauc-Message: installing /media/usb/up2.raucb:  20% Verifying signature
rauc-Message: installing /media/usb/up2.raucb:  40% Verifying signature done.
rauc-Message: installing /media/usb/up2.raucb:  40% Checking bundle done.
rauc-Message: installing /media/usb/up2.raucb:  40% Verifying manifest
rauc-Message: installing /media/usb/up2.raucb:  40% Loading manifest file
rauc-Message: installing /media/usb/up2.raucb:  50% Loading manifest file done.
rauc-Message: installing /media/usb/up2.raucb:  50% Verifying manifest checksums
rauc-Message: installing /media/usb/up2.raucb:  60% Verifying manifest checksums done.
rauc-Message: installing /media/usb/up2.raucb:  60% Verifying manifest done.
rauc-Message: installing /media/usb/up2.raucb:  60% Determining target install group
rauc-Message: installing /media/usb/up2.raucb:  80% Determining target install group done.
rauc-Message: installing /media/usb/up2.raucb: 100% Installing failed.
rauc-Message: installing /media/usb/up2.raucb: LastError: Handler error: Failed marking slot rootfs.1 non-bootable
Installing `/media/usb/up2.raucb` failed

root@arm:~# tail -f --lines=20 /var/log/syslog 
.
.
.
Sep 15 12:16:14 arm rauc[5607]: Using default handler
Sep 15 12:16:15 arm rauc[5607]: rauc-Message: Marking target slot as non-bootable...
Sep 15 12:16:15 arm rauc[5607]: Cannot parse config file: No such file or directory
Sep 15 12:16:15 arm rauc[5607]: Error: environment not initialized
Sep 15 12:16:15 arm rauc[5607]: (rauc:5607): rauc-WARNING **: fw_setenv failed: Child process exited with code 1
Sep 15 12:16:15 arm rauc[5607]: (rauc:5607): rauc-WARNING **: failed marking as bad
Sep 15 12:16:16 arm rauc[5607]: (rauc:5607): rauc-WARNING **: Handler error: Failed marking slot rootfs.1 non-bootable
Sep 15 12:16:16 arm rauc[5607]: rauc-Message: installing /media/usb/up2.raucb: Handler error: Failed marking slot rootfs.1 non-bootable
Sep 15 12:16:16 arm rauc[5607]: rauc-Message: installing /media/usb/up2.raucb: finished
Sep 15 12:16:16 arm rauc[5607]: rauc-Message: installing `/media/usb/up2.raucb` failed: 1

Any idea on which configuration needs to done. The error message is not giving sufficient information.

ejoerns, 2017-09-15:
IMHO the log message gives sufficient information. It says

> Sep 15 12:16:15 arm rauc[5607]: Cannot parse config file: No such file or directory
> Sep 15 12:16:15 arm rauc[5607]: Error: environment not initialized

Did you set up the U-Boot environment configuration `/etc/fw_env.config` ?
https://www.denx.de/wiki/DULG/HowCanIAccessUBootEnvironmentVariablesInLinux

Maybe we should extend the documentation concerning this point

ejoerns, 2017-09-18:
I've made PR #174  with extended documentation on how to interact with U-Boot. Please check if it also helps you guiding to the right direction concerning your issues.

aapurohit, 2017-09-19:
yes thank you working on it

closed 2017-09-19 by aapurohit