rauc/rauc #176: problem integrating rauc hawkbit

aapurohit, 2017-09-25:
I have installed rauc-hawkbit client on my beaglebone.
My configuration file is :
[client]
hawkbit_server = 192.168.0.15:8080
ssl = false
ca_file =
tenant_id = DEFAULT
target_name = test
auth_token = ahVahL1Il1shie2aj2poojeChee6ahShu
mac_address = ff:ff:ff:ff:ff:ff
bundle_download_location = /tmp/bundle.raucb
log_level = debug

my hawkbit server is running on my local machine who's ip and port is mentioned at hawkbit server.
beaglbone and server are in the same LAN .

on running rauc-hawkbit-client on beaglebone:
DEBUG Using selector: EpollSelector
DEBUG GET http://192.168.0.15:8080//DEFAULT/controller/v1/test
WARNING Polling failed due to TimeoutError
INFO Retry will happen in 60 seconds

Strangely, this error is occuring only on my beaglebone.

I tried installing rauc-hawkbit on another pc and pc-pc communication is flawless.
Any idea on where am I going wrong or what other things do I need to check?

P.S. : I wrote a sample code only sending GET request from my beaglebone to hawkbit server. That code is working fine as I am getting 200 OK and config info from the server in the sample code output.

BastianStender, 2017-09-25:
Hi,

maybe you are actually running into the timeout as the Hawkbit server on your Beaglebone cannot answer the request in time?

If you notice long response times try increasing the timeout in https://github.com/rauc/rauc-hawkbit/blob/master/rauc_hawkbit/rauc_dbus_ddi_client.py#L34 with something like:
`self.ddi = DDIClient(session, host, ssl, auth_token, tenant_id, target_name, timeout=20)`

Bastian

BastianStender, 2017-09-25:
This should be closed because the right place to discuss this is the issue in https://github.com/rauc/rauc-hawkbit/issues/2.

ejoerns, 2017-09-25:
I agree with @BastianStender. Double posting definitely won't speed up issue investigation on our side. I will close this one as the question is more related to rauc-hawkbit.

closed 2017-09-25 by ejoerns