rauc/rauc #184: rauc install does not work with barebox and the bootloader spec

mischelebuha, 2017-10-27:
I use barebox as bootloader with following functionality:
- bootchooser
- state framework
- bootloader spec
- nand/mtd/ubi/ubifs

In the bootloader spec i use the option "linux-appendroot true". Thus barebox only knows the boot medium and the boot partition. All other information, such as the position of the kernel and the device tree, will be completely described in the rootfs. Everything works fine with the sd-card (boot mmc0.0) and via nfs (boot nfs).

Now i setup the nand with ubi and after the start of linux (**boot nand0.root.ubi.system0**) the command-line looks like this:
`root=ubi0:system0 ubi.mtd=root rootfstype=ubifs console=ttymxc0,115200n8`

It works as described in the documentation of Barebox:
>For UBIFS fileystems it will be **root=ubi0:volname** ubi.mtd=mtdpartname rootfstype=ubifs. NFS filesystems will result in root=/dev/nfs nfsroot=ip:/path/to/nfsroot,v3,tcp. The v3,tcp part is configurable in global.linux.rootnfsopts.

The rauc system.conf:
>bootloader=barebox
>
>[slot.rootfs.0]
>device=/dev/ubi0_0
>type=ubifs
>**bootname=ubi0:system0**
>
>[slot.rootfs.1]
>device=/dev/ubi0_1
>type=ubifs
>**bootname=ubi0:system1**

So the "bootsname" given in the command line can be matched.

But if I want to install an update with rauc, I get the following error:

>rauc[346]: rootfs -> rootfs.0
>rauc[346]: Using default handler
>rauc-Message: Marking target slot as non-bootable...
>rauc[346]: Failed to set variable **bootstate.ubi0:system0.priority** in state state to 0: No such file or directory

These errors occur because there is no entry "ubi0:system0" in the device tree. Now I could change the entry in the device tree. However, in the label of a devicetree entry is the sign "**:**" forbidden, or not possible. Am I the only one that has this constellation?

And how could you fix this? I would reluctantly give up the bootloader spec!

ejoerns, 2017-10-27:
Booting with bootspec is definitely the preferred way we use, too, if possible! You will not have to give it up.

Well, you actually have to set the `bootchooser` boot target as the `bootname` in order to get this working, not what in `root=` is.
The bootname enables RAUC to interact with the boot selection logic, the barebox bootchooser in this case. Thus it must know about the names bootchooser and state use, not from which ubi volume we are booting.
Telling where we are booted from will then be done by additional kernel commandline parameters passed by bootchooser.

The [Barebox Integration](http://rauc.readthedocs.io/en/latest/integration.html#barebox) chapter in the documentation tells you a bit more about how to configure bootchooser with RAUC.

mischelebuha, 2017-11-02:
Thanks for the help the problem has been solved.

In Germany one would say "I stood on the hose". :-D

I manually started the boot process in all attempts (**boot nand0.root.ubi.system0**). Therefore, the variable "**bootchooser.active=**" was not generated. And the resulting "error messages" did not help me either. Maybe the missing variable "bootchooser.active" should be mentioned in the error message.

Is it possible to start the bootchooser manually with a specific target (without changing the priority)?
For example, **boot bootchooser -t system0** ?

ejoerns, 2017-11-02:
Auf dem Schlauch stehen wir alle mal ;)

Well, the missing variable is sufficient but not necessary. RAUC also accepts ``bootstate.active=`` (for compatibility reasons), the generic ``rauc.slot=`` an is also able to handle simple ``root=`` options matching the 'bootname'. Thus it is not that easy to emit a helpful error messages.

No, manually selecting a boot target without modifying prios is not possible at the moment.

For both points mentioned, patches improving the current situation are always welcome ;)

closed 2017-11-02 by mischelebuha