rauc/rauc #185: Missing information in the documentation

mischelebuha, 2017-11-08:
Missing information in the documentation

I'm experimenting with RAUC and Barebox right now. At the beginning I packed my RFS as tar.gz into a RAUC bundle. Now, a NAND is used as memory. Now a UBI file system is needed. My RAUC bundle now contains UBIFS image instead of a tar.gz.

The documentation contains only this chapter:
[2.7. Installation and Storage Handling](http://rauc.readthedocs.io/en/latest/basic.html#installation-and-storage-handling)

However, it does not show that a RAUC bundle with a UBIFS image uses UBIVOL (type = ubivol) in the system.conf.

Here you should add even more documentation, that other people do not waste time.

ejoerns, 2017-11-08:
Is this the information you require?
http://rauc.readthedocs.io/en/latest/integration.html?highlight=ubivol#slot-type

As noted there, you may use tar but you have to specify the slot type to be `ubifs`. The difference to `ubivol` is that `ubivol` does not pin the file system to use, thus you could also place a squashfs in here. With `ubifs` you pin the file system to use to `ubifs`. Now RAUC can safely call `mkfs.ubifs` on this volume in case you provide a tar archive instead of an ubifs image.

ejoerns, 2017-11-20:
Did that help you? Are you missing something? Otherwise I would close this one.

mischelebuha, 2017-11-27:
Yes, that helped me. Maybe you could make the chart clear, so that others do not come across this problem. In fact, only the column or the label "system.conf" is confused.

ejoerns, 2017-12-04:
I assumed the chart would point out that you can combine both tar.gz and ubifs image type with ubifs image type. What do you find confusing concerning the label `system.conf`. Would adding something like 'slot type' make it more clear?
From my point of view it would also be helpful linking the chapter containing the full matching table to the 'Installation and Storage Handling' chapter. Would you agree?

dhr2, 2017-12-26:
Hi,
I am also facing issue while installing RAUC bundle. This bundle has following system.conf file. 
```
[slot.RFS1.0]
device=/dev/mtd1
type=ubifs
```

and I packed my RFS1 as tar.bz2 into a RAUC bundle.
At time of installing RAUC bundle, gives me below error.
"
**(rauc:691): rauc-WARNING **: No target for class 'RFS1' found!

(rauc:691): rauc-WARNING **: Could not determine target group**
"
Help me to identify issue in system.conf file.

ejoerns, 2018-01-02:
> This bundle has following system.conf file.

Your bundle does not have a system.conf file, it has a Manifest instead. Thy system.conf must be in the rootfs of your target you intend to update.

Please show the entire system.conf (at least the slots defined) of your system. The except above cannot describe a redundant rootfs system.

ejoerns, 2018-01-16:
Any news on this? Otherwise I would like to close this one.

dhr2, 2018-01-16:
Hello,

Sorry for late reply. Yeah got the solution. There was mismatched between my Manifest and system.conf file. 
Thanks for your reply :)

ejoerns, 2018-01-16:
Ok, thanks for feedback!

closed 2018-01-16 by ejoerns