rauc/rauc #190: Do not skip a slot even if its hash matches (with hooks)

jktjkt, 2017-11-28:
This is with RAUC 0.1.1.

I've got an A/B rootfs and A/B persistent storage mounted at /cfg (I want to guard against the persistent storage failing to mount). When doing an update, I want to handle the rootfs as usual, and I want my new `/cfg` to contain the same files as the current one.

Right now I have an ugly hack as a custom hook for this. That hook copies data around, and it mostly works -- except when the hash of the old tarball for `/cfg` happens to match the current one. In that case, my hook doesn't run, and on next reboot, my `/cfg` data are not taking an effect.

I understand the motivation, perhaps I should not provide a tarball for my `/cfg` in the first place. Pointers in that direction are welcome. Here's what I would like to have:

- no change for `rootfs`
- re-mkfs the new `/cfg`, always, unconditionally
- run a hook/script/whatever, always, unconditionally

Can I do that with RAUC?

Here's my `system.conf`:

```
[system]
compatible=czechlight-clearfog
bootloader=uboot

[keyring]
path=/etc/rauc/keyring.pem

[slot.rootfs.0]
device=/dev/mmcblk0p1
type=ext4
bootname=A

[slot.cfg.0]
device=/dev/mmcblk0p2
type=ext4
parent=rootfs.0

[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=B

[slot.cfg.1]
device=/dev/mmcblk0p4
type=ext4
parent=rootfs.1
```

My `manifest.raucm` looks like this:
```
[update]
compatible=czechlight-clearfog
version=dev

[hooks]
filename=hook.sh

[image.rootfs]
sha256=d12933aa8695030615483aa1ce03ea8ef83232355109244bbb6fc653c050a896
size=15241044
filename=rootfs.tar.xz

[image.cfg]
sha256=5ca1b6c461fc194055d52b181f57c63dc1d34c19d041f6395e6f6abc039692bb
size=108
filename=cfg.tar.xz
hooks=post-install;
```

And here's the `hook.sh`:
```
#!/bin/sh

case "$1" in
  slot-post-install)
    case "$RAUC_SLOT_CLASS" in
      cfg)
        if [[ -d /cfg/etc ]]; then
          cp -a /cfg/etc ${RAUC_SLOT_MOUNT_POINT}/
        fi
        ;;
      *)
        echo "Internal error: hook mismatched"
        exit 11
    esac
    ;;
  *)
    echo "Internal error: unrecognized hook"
    exit 11
    ;;
esac

exit 0
```

ejoerns, 2017-11-29:
Did I get it right that you intend to populate the /cfg with the tarball content and *additionally* want to copy data from your current system? Or is the tar just used as a hook for actually triggering slot update?

If you only intend to copy data for this slot, you could assure to have mounted the 'other' cfg partition, too and only specify a rootfs image and run a post-install hook for it that copies content from /cfg to the other slots mount point.
Otherwise maybe a mechanism as proposed in #181 would be what you require?

ejoerns, 2017-12-06:
Any updates on this?

jktjkt, 2017-12-06:
The fake tarball for `/cfg` is really just a trigger for me. It's an empty tarball, after all, and I'm using it just for triggering my hook script. A benefit of that is that RAUC has already mounted a correct partition for me, and that I don't have to figure out the correct block device to use.

Based on a quick glance, #181 looks exactly like what I'm looking for.

Sorry for a late response.

ejoerns, 2018-01-08:
Thanks for the udpate. #181 just got merged, so I will close this.

closed 2018-01-08 by ejoerns