rauc/rauc #192: Slot hook does not execute pre-install

Blackrose, 2017-12-02:
Hi, I use RAUC to update uboot into emmc flash.Before start, it needs to some necessary operates.So, I add 'pre-install' hook to handle that.Below is the content of manifest.raucm.
```
    [hooks]
    filename=hook.sh

    [image.bootloader]
    filename=u-boot-emmc.imx.img
    hooks=pre-install;install;

    [image.core]
    filename=boot.vfat.img
    hooks=post-install;
```
And below is the content of hook.sh.
```
    echo "**** is $1"
    case "$1" in
    slot-pre-install)
        echo "pre device is $RAUC_SLOT_DEVICE"
        ;;

    slot-install)
        echo "install device is $RAUC_SLOT_DEVICE"
        ;;
    slot-post-install)
        echo "pre device is $RAUC_SLOT_DEVICE"
        ;;
    *)
        echo "Hook arg is $1"
        exit 1
        ;;
    esac
```
When the update running, I just saw slot-install and slot-post-install be executed.The result shows me the RAUC not pass 'slot-pre-install' to hook.sh file.
Does the RAUC can not support it?

ejoerns, 2017-12-03:
As written in the documentation, pre-install hooks are supported by RAUC.

To find out why it does not seem to work in your case, it would be helpful to have the slot configuration and a log file of the RAUC install that fails to run the pre-installer.

Note that if you run the above hook file, you will not be able to differenciate between the pre-install and post-install hook, because they print the exactly same message (`pre device`) ;)

Blackrose, 2017-12-04:
I had fixed the print message.And the first line print the $1 argument, it only displays slot-install and slot-post-install.I will view log to find out the reason.

Blackrose, 2017-12-09:
I changed bootloader slot use 'pre-install' hook only.It's work well.While the 'pre-install' cannot work with 'install' hook set in the slot.

ejoerns, 2017-12-09:
Oh yes, you are totally right. Seems I didn't recognize that when scanning your example.
But, it is clearly noted in the documentation (http://rauc.readthedocs.io/en/latest/using.html#slot-hooks, section 'Install Hook'.

> The install hook will replace the entire default installation process for the target slot of the image it was specified for. Note that when having the install hook enabled, pre- and post-install hooks will not be executed.

Thus, the valid solution if you want a totally custom installation would be to use the 'install' hook and do it all from there manually, or, if you want the default installer to handle the image copying, then only use the pre-install hook as you did.

Blackrose, 2017-12-10:
Thanks again.I will try to use 'install' hook.

closed 2017-12-10 by Blackrose