rauc/rauc #212: meta-rauc: run with --debug

anguslees, 2018-01-14:
rauc doesn't give a lot of information by default, and basic things like "which files were included" can be hard to work out.

Suggestion: Add `--debug` to the `rauc` command in `bundle.bbclass`.  Bitbake logs output to files, so this won't affect regular (successful) use, and will greatly aid someone who is trying to debug an issue.

ejoerns, 2018-01-15:
Thanks for reporting, I guess this is a useful hint!

Would you like to send a patch for this to meta-rauc? Please also consider placing meta-rauc-related questions directly as issues for [meta-rauc](https://github.com/rauc/meta-rauc) for the next time. It's more targeted and eases linking between PR and Issue :)

ejoerns, 2018-01-24:
As I just touched meta-rauc for a different PR I simply did the addition you requested and pushed as PR [#32](https://github.com/rauc/meta-rauc/pull/32) to meta-rauc repo.

closed 2018-01-24 by ejoerns