rauc/rauc #219: Cross-Compile in host platform

aobalioglu, 2018-01-23:
Hello,

I compiled Rauc for my host machine without any problems. 

On my target machine (AM3354, Kernel 3.14, Custom Rootfs), a rootfs that I compiled with busybox is running and and no any compiler tools included. Can I compile Rauc for target machine with cross compiler on my host machine? I did not see a help file on this topic. 

Thank you.

ejoerns, 2018-01-24:
Well, cross compiling RAUC is a standard use case.

May I ask how you build you rootfs? In general the best way to build an embedded linux is using an embedded Linux build system such as Yocto/OE, PTXdist or Buildroot. There you can have git-versioned reproducible builds of you entire system.

Support for RAUC is included in all these tools mentioned above. And RAUC comes with standard autotools build system. Thus if you intend to cross-compile it manually, do it is just the way you cross-compile your other applications.

https://www.gnu.org/software/automake/manual/html_node/Cross_002dCompilation.html

ejoerns, 2018-02-07:
Did that help you?

closed 2018-02-21 by ejoerns

IvanSlaev, 2020-11-25:
@ejoerns, hello, I am here a bit late but after I experience a really confusing behavior of rauc when starting from systemd service and starting it by hand - when by hand I was able to install update but when with systemd I get an error telling me it could not mount, exiting with "No such file or directory". 
Anyways, I found your answer very unsatisfying as you do not mention that cross compiling takes a lot of work supplying the compilation with all the needed libraries and on top of it, they should be prebuild for the --host. The configuration is only checking and warning about the missing libraries.
Is there something which can be done about it?

ejoerns, 2020-11-27:
I'd always suggest using an embedded Linux build system for building an embedded Linux system. Good choices are Yocto, PTXdist, Buildroot, which have all the dependency handling, sysroot setup etc. abstracted that you need for cross-compiling.
How do you assemble your system?
RAUC is itself well-prepared for being cross-compiled to the best of our knowledge, but solving the general challenges of cross-compiling is surely out-of-scope for RAUC. 

Maybe, for the original problem, it could be an option to post the actual log message of the issue you encounter, probably with the configuration used, etc.