rauc/rauc #221: unintuitive option `ignore-checksum`

ukleinek, 2018-01-26:
IMHO the slot option `ignore-checksum` is misnamed. For a piece of software that should write an image somewhere I would expect that adding `ignore-checksum` results in the installation of the image even if the checksum that is part of the image doesn't match. According to [its documentation](http://rauc.readthedocs.io/en/latest/reference.html?highlight=ignore-checksum#slot-slot-class-idx-section) its about skipping installation if the image to install is already installed.

IMHO a better name for this function would be `force-install-same`.

jluebbe, 2018-02-01:
@jremmet what do you think about renaming this to `force-install-same`?

jremmet, 2018-02-02:
Agree from user view `force-install-same` is much better

ejoerns, 2018-02-21:
@ukleinek would you like to send a patch? For backwards-compatibility we must accept both options.

ejoerns, 2018-06-08:
Anyone who likes to implement this? ;) Or should be close it as unresolved?

jluebbe, 2018-06-08:
I agree that `force-install-same` is much clearer. `ignore-checksum` needs to stay for compatibility.

jremmet, 2018-06-18:
I'll look into it.

ejoerns, 2018-07-05:
This was fixed by #280. Thanks!

closed 2018-07-30 by ejoerns