rauc/rauc #226: Segmentation fault when running rauc on the target system

nettercm, 2018-02-09:
Was trying to do a quick sanity check.

When I run rauc on my target system, I get a segmentation fault.  What could cause this?  The exact same command, w/ exact same input files runs fine on the host.

I built rauc v0.3 via buildroot.  (I manually changed the buildroot package settings so that rauc v0.3 gets used instead of 0.2)

My arm-linux-gnueabihf target does not have systemd or D-Bus.  Could that be an issue?

Attached are the files used for running the command
[files.zip](https://github.com/rauc/rauc/files/1712381/files.zip)

# rauc --keyring=./demo.cert.pem info update-2015.04-1.raucb
Invalid byte sequence in conversion input

(rauc:8680): GLib-CRITICAL **: g_error_new_literal: assertion 'message != NULL' failed

(rauc:8680): GLib-CRITICAL **: g_propagate_error: assertion 'src != NULL' failed
Segmentation fault

ejoerns, 2018-02-09:
Looks like the exact same symptoms we see in #224.

nettercm, 2018-02-09:
Based on comments made in #224 , I tried a different version of glib and I was able to run rauc on my target.  Basically, I grabbed some pre-built binaries from debian armhf (glib2.0 2.50.3,  plus some dependencies it wanted such as libpcre3_8.39 and libselinux1_2.6).  So maybe what I should do is switch from the latest buildroot to a stable or LTS version.  The reason I used the latest version of buildroot is because only that contain the rauc packages.

jktjkt, 2018-02-10:
Alternatively, you can revert rauc to v0.1.1 which works even with the latest version of libglib2 as shipped in buildroot master. That's what I did in the meanwhile.

ejoerns, 2018-02-10:
I assume this to be a follow-up issue triggered by an error during rauc install, thus it would be worth testing if willfully triggering a similar error in RAUC 0.1.1 (or 0.2) will really not lead to issues with this glib version.

ejoerns, 2018-02-13:
I will close this as duplicate of #224 to focus the discussion on that one.

closed 2018-02-13 by ejoerns