rauc/rauc #227: Is D-Bus support on the target required?

nettercm, 2018-02-09:
Based on the output below, it looks like the answer is "by default, yes", but can the D-Bus dependency be eliminated?

# ./rauc -d --conf=rauc.config --keyring=./demo.cert.pem install  update-2015.04
-1.raucb
Domains: 'rauc'
(rauc:11940): rauc-DEBUG: install started
(rauc:11940): rauc-DEBUG: input bundle: /home/root/rauc/update-2015.04-1.raucb
Error creating proxy: Could not connect: No such file or directory
D-Bus error while installing `/home/root/rauc/update-2015.04-1.raucb`

ejoerns, 2018-02-09:
We have the option to configure RAUC with `--disable-service` which will replace the D-Bus client calls of the RAUC command line tool to the RAUC service by direct method calls in RAUC without an extra background service.
Note that this feature is not heavily tested by default.

ejoerns, 2018-02-16:
Did that help you and answer your question?

kgolding, 2018-06-19:
I can't find any documentation on the `--disable-service` option. Where/how is this used please?

ejoerns, 2018-06-21:
@kgolding Well, the 'how' is basically what I described above. About the where: Well, basically in systems where you don't have D-Bus or don't want to use it. Probably a minimal intramfs I can guess as a possible use case. To be hones I haven't had more than once case where this was actually used (and I know about) so far.

What is your motivation on looking for this option, for example?

ejoerns, 2018-08-13:
To close this, I've added #304.
@kgolding feel free to have a look and let us know if that would have helped you.

PR #304 (2018-08-20): docs: add minimal hint on --disable-service in RAUC faq
The ability to use RAUC in non-D-Bus mode is not very well documented.
Thus we add at least a little hint in the FAQ to mention it despite we
do not want to encourage people to use RAUC that way if not mandatory.

Fixes #227
commit 1b5dfff2 (2018-08-20): docs: add minimal hint on --disable-service in RAUC faq

The ability to use RAUC in non-D-Bus mode is not very well documented.
Thus we add at least a little hint in the FAQ to mention it despite we
do not want to encourage people to use RAUC that way if not mandatory.

Fixes #227
closed 2018-08-20 by jluebbe