rauc/rauc #236: Multiple compatible strings in manifest file

sloe10, 2018-02-23:
Is it possible to compare compatible string against array of strings during update? It would be very nice feature to have it.

ejoerns, 2018-02-23:
Yes and no. :)

Well, the default mechanism intentionally only allows a single compatible string to make sure the bundle is for the correct device (or group of devices!) you intend it to be for.

But, there is support for 'variants' (see https://github.com/rauc/rauc/blob/master/docs/advanced.rst#handling-board-variants-with-a-single-bundle) that allows you to handle different variants of a board (either by using dtb compatibles, or setting explicit names) with the same bundle. This works on the image level which means that you can for example include seperate bootloader images for each variant of a board while having a single rootfs image supporting all varariants.

If that does not match your needs, there is still the possibility to place an 'install-check' hook in your bundle that replaces the entire default compatible comparison by a custom script or binary. Refer http://rauc.readthedocs.io/en/latest/using.html#install-hooks for more details.

ejoerns, 2018-03-08:
Did that answer your question?

sloe10, 2018-03-13:
Yes, it helped me. Thank you

closed 2018-03-13 by sloe10