rauc/rauc #241: UBIFS type and determinate slot

wlot, 2018-03-08:
Hi,
we use UBIFS on our device. For kernel boot args we set
`ubi.mtd=8 root=ubi0:rootfs0 rootfstype=ubifs` to mount _rootfs0_ and
`ubi.mtd=8 root=ubi0:rootfs1 rootfstype=ubifs` to mount _rootfs1_.

in `/etc/rauc/system.conf` we create two sections:
```
[...]

[slot.rootfs.0]
device=ubi0:rootfs0
type=ubifs
bootname=system0

[slot.rootfs.1]
device=ubi0.rootfs1
type=ubifs
bootname=system1

[...]
```
RAUC recognize booted slot, because of mount (ubi0:rootfs0 -> /), but if we try to install a bundle it fails.
rauc try to open file like: `/etc/rauc/ubi0:rootfs0`. This file is not exists of course.

Have I made a mistake somewhere ?
if not, is it possible to add a support for ubifs type `ubiX:VOLNAME` syntax?

Regards,
Andreas

ejoerns, 2018-03-08:
Hi Andreas,

two things: First one is that RAUC indeed is not yet able to handle this kind of syntax, second is that I'm a bit confused by the resulting issue.. 

But let me be a bit more detailed about the first point: The `device` parameter should be something like `/dev/ubi0_0` to make the update work with the current setup as RAUC will use this as the target file for writing the image to.

The way you can bridge the gap from the bootloader to the ubi then is setting e.g. `rauc.slot=system0` from the bootloader when booting system0. This is anyway the more explicit and recommended way to interact with the bootloader compared to the potentially more fragile detection of the mounted root.

One could talk about adding the `ubi0:rootfN` syntax as an alternative but I would not recommend this as the primary solution.
May i ask which bootloader you use?

Anyway, the second thing you noted, that it tries to access `/etc/rauc/ubi0:rootfs0` sounds a bit confusing. Could you provide a log for that? Just want to make sure we do not run into potentially severe problems here...

hamster170780, 2018-03-08:
Hi Enrico,
we are using `barebox` as bootloader.

Thank you for your support!
So I will add boot args `rauc.slot=systemX` for *booted slot* recognition and set device file to `/dev/ubi0_X`. 
Maybe it is helpful for other people if a note will be add to documentation. UBIFS docu describes both ways for mount ubsfs.

The logs conjunction to `/etc/rauc/ubi0:rootfs0` I will create tomorrow. I guess `/etc/rauc` prefix appears  because `device=ubi0:rootfs0` is relative (without / at the beginning) and our `system.conf` is stored at `/etc/rauc`.

Regards,
Andreas

ejoerns, 2018-03-08:
Hi Andreas,

oh, yes you're right about the relative path. So no need for a log here. I was just a bit too paranoid ;)

If you use the barebox bootloader, I assume you use the bootchooser framework for booting? This already sets the `bootchooser.active=<bootname>` kernel parameter which is equivalent to `rauc.slot=<bootname>` for RAUC. Thus you could simply use your setup as it is and would not have to change anything.

Best regards, Enrico

hamster170780, 2018-03-08:
Hi Enrico,
yes, we use bootchooser framework and `bootchooser.active=<bootname>` I've seen on kernel logs.
Was this support by rauc 0.3?

Good night,
Andreas

ejoerns, 2018-03-08:
Nope, a bit before ;) https://github.com/rauc/rauc/commit/a35d58aa7a565ef25185589ab82156a3bf7ece5d

Good night! Enrico

closed 2018-03-26 by ejoerns