rauc/rauc #244: casync convert documentation, missing args

potens1, 2018-03-09:
Hi,

the documentation states we can simply call

 `rauc convert conventional-bundle.raucb casync-bundle.raucb`

 to create a casync bundle, but it seems the `--cert`, `--key` and `--keyring` parameters are required for this to work since
```
$ rauc convert conventional-bundle.raucb casync-bundle.raucb
Cert and key files must be provided
```
and
```
$ rauc --cert=mycert.cert.pem --key=mycert.pem convert conventional-bundle.raucb casync-bundle.raucb
No keyring file provided
```

potens1, 2018-03-09:
Yeah, for sure, I'll do a PR.

(in fact, I started to do it, but after, I was wondering if it was really a doc problem)

That's something a bit difficult when evaluating rauc, there are not a lot of _complete_ exemple (understand, cut, paste, check if that worked, and then when you have an end to end exemple working, start to change things)

ejoerns, 2018-03-09:
Thanks for your contribution!

Doing 'complete' examples is a little tricky as most things depend on the individual system requirements but I agree that more examples would make it more easily to figure out what to do.

If you've got any good proposal for this, please let us know.

closed 2018-04-10 by ejoerns