rauc/rauc #248: Improved D-Bus Interface

ejoerns, 2018-03-14:
I would like to start this as a discussion base for future extensions, and possible re-designs of the current. D-Bus API. Thus if you've got any need for functionality, any concern about the existing implementation, or similar, please note it here so we can collect it.

One initial discussion base we already had in #238 were the current limitation of the `Info` D-Bus method was mentioned. Our current approach to an extensible and generic interface is to have a pure string-variant-dict-based approach. Similar to what we have in the `Status` method.
A drawback of this might be that this way is "too flexible" and not really an API anymore..

Another limitation we already had issues with in the past is the curent `Install` method that does not allow to provide further options and thus required us to move the 'delayed activation' feature (https://github.com/rauc/rauc/pull/134/commits/35a13bc201ca4709ded80e856eb5a51e31fbb6a5) to a system.conf-only feature, for example.

embeddedrocks, 2018-10-12:
Extending the ```Info``` method with a ```no-verify``` attribute would be great (as you already do via command line).
Background: We are using a USB pen drive to check for update files. The drive can contain multiple files. It can take a great amount of time validating each file. So currently the files are not validated when displayed, but of course will be later on. But in order to do so the command line is used instead of DBUS interface.

ejoerns, 2018-10-12:
@embeddedrocks thanks for the hint!

To be honest, the Info 'API' is rather limited and not actively propagated by us as we know it lacks a lot of features. The command line interface does not interact with it but inspects the bundle directly. I am not fully sure if this should be changed in the future but one may provide a better Info API anyway.

ejoerns, 2018-10-12:
I've prepared an initial guess on how a new interfaced could look like some time ago. I'll just dump it here for reference. I fear it lacks at least a parameter for 'Info' ;)

```
/de/pengutronix
/de/pengutronix/Rauc1
  > Interfaces
    > de.pengutronix.Rauc1
      > Methods
        - Install(String bundle, Dict of {String, Variant} options)
        - Info (String bundle) ->
        - GetAllSlots() -> Array of[Object Path]
        - GetBootedSlot() -> Object Path
      > Properties
        - String Compatible
        - String Variant
        - Struct of (int32, String, int32) Progress
        - ? LastError
    > de.pengutronix.Rauc1.Bootloader
      > Methods
        - MarkGood(String slot)
        - MarkBad(String slot)
        - MarkActive(String slot)
/de/pengutronix/Rauc1/Slot/0
  > Interfaces
    > de.pengutronix.Rauc1.SlotStatus
      > Methods
        - Write (String image)
      > Properties
        - String Description
        - String MountPoint
        - String State
        - Object Path Parent
    > de.pengutronix.Rauc1.Installation
      > Properties
        - String Compatible
        - String Version
        - String Build
        - String ActivatedTimestamp
        - Integer ActivatedCount
        - String UpdatedTimestamp
        - Integer UpdatedCount
/de/pengutronix/Rauc1/Slot/1
  ...
/de/pengutronix/Rauc1/Slot/2
  ...
/de/pengutronix/Rauc1/Slot/3
  ... 
```

embeddedrocks, 2018-10-12:
Many thanks for your thoughts @ejoerns 

I guess I just missed to mention another import part why we use the ```Info``` command :-)
To make sure the RAUC packet is compatible to our system we included further information in the ```version``` field (beside ```compatible```, ```describtion```, etc.). Prior to any installation we need to get hold of this information. The only chance to do so, according to my knowledge, is using the ```Info``` command.

xyzzy42, 2018-12-19:
I'd like to be able to query rauc to determine:
1. Which slot we are running now.
2. Which slot we will be running on reboot.
3. Which slot will be replaced if an update is done.

You can get 2 via BootSlot, and 1 (in a different format) via GetPrimary, but 3 is only there by assuming it is the slot which is not returned by BootSlot.

xyzzy42, 2018-12-19:
We have product requirements to display the currently installed firmware in each slot, allow switching between slots, and show what firmware will be replaced when an update is applied.  I can't ask rauc what it will do when it installs a bundle, but rather must independently arrive at the same answer rauc will.

There is also the lack of orthogonality, in that booted slot and active slot aren't returned in the same way by the dbus interface.  The booted slot returns the "bootname" but the method for active slot returns the slot name, e.g. "A" vs "rootfs.0" for the same slot.

zonque, 2019-02-05:
As somebody who uses the DBus API in a product I'd like to know whether you intent to keep the current interface in place?

ejoerns, 2019-02-05:
@zonque Of course. This is a non-internal API and thus must be kept stable for backwards compatibility. We do not intentionally break existing setup ;)

The way to go would be to let the two API's exist in parallel. This of course adds some overhead but is unavoidable.

It could be possible that we deprecate it some day. But only in a very far future and not without a major version change of RAUC. But even in this case, it could be still possible to update your device in a two-step process and adaption of userspace code.

zonque, 2019-02-06:
Nice, thanks for the answer. FTR, I've written a Golang package to interface with that API and published it here: https://github.com/holoplot/go-rauc

Once the new API is in place, I can extend that package to support the new scheme.

ejoerns, 2019-03-21:
@zonque Cool! Thanks for referencing this.

We could also collect existing bindings in the documentation if you like to make a PR.

ejoerns, 2019-06-06:
FTR: With #446 we started targeting the most important limitation of the current API by introducing an alternative install method being capable of handling options.

jonaskgandersson, 2019-08-11:
I an looking into an way to use RAUC host tools for creating and resigning bundles from an third party tool, maybe creating an GUI tool for it. An d-bus interface could be an way, do you have any experience in something similar?

ejoerns, 2019-10-05:
@jonaskgandersson Sorry for answering a bit late: You mean experience in creating a host GUI? Or how to have a D-Bus interface for the bundle signing?

To be honest, at the moment the focus of D-Bus interfacing was more on the service side, i.e. what is done on the target. Thus there currently is no interface that allows creating bundles. And the interface for inspecting them is quite limited as noted above.

Thus maybe host operations are something we also should consider. Do you have any proposals?

jonaskgandersson, 2019-10-29:
If you break it down It s more of an question of host operations as you already have pinned down.

Our need was/is to do bundle operations like signing and verification without the knowledge or access to our build system (Yocto). We have an need for an "easy" host tool for staff that have no RAUC knowledge. The D-Bus interface was only seen as an possible interface for an custom application, maybe some form of web app.

Jo3Zhou, 2024-09-20:
Hi all,

Thank you for the great work on this project.

I noticed that the gdbus-codegen tool is used to generate D-Bus APIs, as seen in the top-level meson.build file with the line: `dbus_sources = gnome.gdbus_codegen`.

I would like to suggest looking into how other open-source projects, such as NetworkManager, handle D-Bus interfaces. These projects, which are also based on GLib, provide a "client library" of APIs directly accessible from C code. This simplifies interaction by abstracting the complexity of D-Bus methods, properties, and signals.

Would it be possible for RAUC to implement something similar? I’d be happy to contribute to this effort if there’s interest.

Looking forward to your thoughts!

nomo-nvm, 2026-09-04:
Also, currently there's no way to listen to slot state changes. Suppose I have a GUI and want to keep it updated when someone calls `mark-bad`. There is really no way of knowing a slot changed state.