rauc/rauc #264: Question: is fixed bundle filename suffix really required?

mhei, 2018-04-09:
> Note that RAUC bundles must always have a `.raucb` file name suffix in order
> to ensure that RAUC treats them as bundles.

When looking at the code, it seems that this is only used to switch between
local filesystem access or network download mode, but there is no other
technical requirement?

So my question is, whether it would be possible to remove this requirement as
I would prefer to distribute the update image files with .image extension.
This way, we could "hide" the internally used update implementation without
renaming the distributed files forth and back. (This is what I would do if you
tell me that this is a fixed design decision)

ejoerns, 2018-04-10:
Well, you already identified the crucial point for this. We currently (still) use it for deciding about using (old) network support or 'conventional' local (and now casync, too) installation.

In case of network mode, one always targets a manifest file iirc. Thus, it might be an option to simply inverse the test by checking if we refer to a manifest (`.raucm` file endling) to use the network mode and use the default mode otherwise.

For now we also have seen this 'explicit' file ending as a feature allowing to prevent form trying to perform useless operations with wrong files (e.g. `rauc install my-rootfs.ext`). But I also know about cases where people created mechanism around RAUC that rename the bundles back and forth to give them their own ending.

From me, I would still prefer `.raucb` as the extension to choose and propagate but I would also consider accepting other extensions. In this case we should at least add a check for the correct 'bundle identiier', i.e. for the bundle to start with a squashfs identifier. I have a patch from our initial casync implementation that actually does this and might be appropriate in a slightly modified way.

@jluebbe what do you think about that topic?

raoulh, 2018-09-07:
I agree with @mhei, forcing to have a .raucb suffix is not really nice. Also based on what you say, doing file detection by checking the suffix is a bad approach. Detecting a type of a file should be done by reading some bytes and checking file headers.

ejoerns, 2018-09-13:
I've made a proposal how we could remove this need, please have a look at PR #328

closed 2018-10-11 by jluebbe