rauc/rauc #265: Filesystem support

aaronovz1, 2018-04-10:
What requirements on the filesystem type does RAUC depend on? Will this work with F2FS?

ejoerns, 2018-04-10:
Hi,

to be honest, I do not know that much about F2FS in detail, because I haven't used it yet. But from a general perspective one way of using RAUC with file systems is to simply give it pre-generated file system images that it writes to a target block device. This case is basically independent from the actual file system used, with 2 exceptions:
* You can only use images with a `*.img` ending (this is because of RAUCs image-specific handling)
* You cannot perform write operations in a post-install hook

Both exceptions will be removable with a simple patch adding an update handler for f2fs, such as those listed here: https://github.com/rauc/rauc/blob/master/src/update_handler.c#L1306.

The other way of using RAUC when dealing with file systems is packaging the target file systems *content* into an archive (e.g. .tar.gz) and let RAUC perform the erasing and formatting of the target slot followed by an archive unpacking. For this it would additionally be required to add support for `mkfs.f2fs` with an extra tar handler, such as we do for ext4: https://github.com/rauc/rauc/blob/master/src/update_handler.c#L879

Hope that helps?

Regards, Enrico

ejoerns, 2018-06-06:
@aaronovz1  Did that help? Are there any questions remaining about this topic from your side?

aaronovz1, 2018-06-06:
I think that provides enough information if someone wanted to add support. This can be closed.

closed 2018-06-06 by aaronovz1