rauc/rauc #266: Updates over http and version control

aaronovz1, 2018-04-10:
Hi,

I am integrating with Yocto and I am trying to wrap my head around how updates and versioning works with RAUC. I am assuming for each new version, RAUC_BUNDLE_VERSION needs to be updated and the *.raucb needs to be uploaded to a web server. I don't see any examples though for how RAUC checks if there is a new version available and downloads/installs it. Is this not built into RAUC? Do I need to use casync along with hawkBit to add this functionality?

ejoerns, 2018-04-11:
Hi,

for what you said about network functionallity basically, yes. RAUC is an update framework focusing on target handling and update artifact (bundle) creation. It provides a D-Bus interface and a command line interface to control it. (see [What is RAUC?](http://rauc.readthedocs.io/en/latest/index.html?highlight=what%20is#what-is-rauc) and 'And What Not' chapter in the documentation). RAUC understands 'here is an update, use it'. Defining where the update comes from (USB stick, deployment server, custom infrastructure..) is highly use-case specific and considered to be out of scope for RAUC.

As you already mentioned with 0.4 RAUC comes with support for casync which does not only provides you network functionalities but also allows chunk-based binary delta updates and potentially allows you to save storage on you server by using a common (deduplicated) chunk store for all updates.

But, this does not solve the question how to trigger an updates. Basically, simple triggering should not be too complicate to implement, depending how 'smart' your infrastracture should be. Even the hawkBit integration is quite easy as it is nothing more than polling on a defined URL, downloading the update artifact (not the actual update data in casync case) if announced, and then triggering RAUC to actually install it. For a first proof-of-concept we have the python-based [rauc-hawkbit](https://github.com/rauc/rauc-hawkbit) client available that is also part of `meta-rauc` Yocto layer. You can use hawkBit without casync if you only care for bundle downloading, i.e. you have enough space on your target and no limited connection.

About the other points:

By default, the `RAUC_BUNDLE_VERSION` is only an informative property and will not be evaluated for installation. RAUC has no concept of versioning only of comparing on equality, which is done based on the calculated image's hashes. Versioning is again very use-case speicifc (scheme, allowing/denying downgrades, further policies, etc.) and  can be implemented using an [install-check](http://rauc.readthedocs.io/en/latest/using.html?highlight=compatible-check#install-hooks) hook for example.

The actual 'versioning' of the updates should be done in your BSP that creates the update bundle. This is the only point that has a well definition about the content of you update.

Best regards, Enrico

aaronovz1, 2018-04-11:
Thanks for the information. Is hawkBit and the rauc-hawkBit client compatible with casync? I am evaluating RAUC as an alternative to SWUpd from Clearlinux which uses binary deltas, so having small minimal delta updates is important.

ejoerns, 2018-04-11:
Haven't tested it yet this way, but basically it should. One would use hawkBit for providing the update bundle that actually only contains the metadata (manifest) and the casync index files for the images used.

Then one would have a separate chunk store that is more or less independent from hawkBit, could be located on the same server or somewhere else.

livioso, 2018-05-24:
@aaronovz1 We've similar constraints and we're considering a similar approach using casync. How did your evaluation go? Any learnings that you'd be willing to share? :)

aaronovz1, 2018-05-24:
@livioso unfortunately due to time constraints our product is going to continue using SWUpd. RAUC and the hawkBit plugin still seem fairly premature and the backend requirements are more complex then our current solution.

ejoerns, 2018-05-24:
@aaronovz1 when you talk about assuming RAUC is 'premature', could you elaborate a little how you come to this conclusion or what gives you that feeling? Would be interesting as a feedback for us.

For rauc-hawkbit I totally agree with you as this was never ment to be used in a product and is just a demo of showing how to do this. It is basically only a bridge between D-Bus and REST API and should be easily replacable by a few weeks of effort.

Thus my question mainly targets RAUC and you experience (e.g. also compared to using SWUpd) with it.

aaronovz1, 2018-05-24:
Sorry, the premature comment was mostly towards the hawkBit integration and the relatively new experimental support for casync. Depending how you interpret version numbering, RAUC and hawkBit have not hit version 1.0 which could suggest alpha/beta stages.

For the most part I like RAUC and also how well documented it is (much better documents then SWUpd). Apart from being time constrained, these couple points below didn't quite fit our product when evaluating a switch from SWUpd:

- Bandwidth constraints. SWUpd does binary deltas, RAUC+casync downloads entire file even if only few bytes changed. Important for 3G/4G and low signal areas.
- RAUC does not have concept of optional SW bundles/packages. In SWUpd we can define for example a "debug" package which is not packaged into the final OS, but can be installed OTA via simple command line and connection to our update server.

ejoerns, 2018-05-24:
Thanks for your feedback!

Yes, the version numbering is something we already discussed, and, yes, we already had the plan to finally make a `1.0` release. But for now, we always found reasons to make with 'the next release' :)
As you know, technicians are always a bit more conservative and want to make a pre-alpha-alpha while the salesman would instantly tag it as 2.0... ;)

Current plan says to tag RAUC as 1.0 when we have collected more experience on the casync feature. It is fully working but might lack some features for other use cases.

> Bandwidth constraints. SWUpd does binary deltas, RAUC+casync downloads entire file even if only few bytes changed. Important for 3G/4G and low signal areas.

I am not sure if I got your objections right, but if a file has changed, RAUC+casync would only need to download the chunk affected, which is potentially smaller then the file itself. One has control over the size of chunks which normally are of a few kB.

> RAUC does not have concept of optional SW bundles/packages. In SWUpd we can define for example a "debug" package which is not packaged into the final OS, but can be installed OTA via simple command line and connection to our update server.

One can not really compare this point as RAUC provides image-based updates and SWupd operates on a file level. The concept of RAUC is to describe a consistent state of the entire system.
For this use-case a corresponding solution would be to use a separate partition for the debug stuff and only include it in a specific bundle or create a full debug rootfs that would be installed in the redundant partition. With casync this will still allow using a lot of chunks from the 'base' system.
But 'package management' is intentionally not covered by RAUC.

The decision if using a file- or image-based approach is a very basic one and mainly driven by the question which parts of your os / hardware you trust and what kind of errors you expect during updating. Thus there might be good reasons to use SWupd instead of RAUC as well as the other way round ;)

aaronovz1, 2018-05-24:
Thanks for the clarification! I guess I misunderstood how casync worked. I thought it just provided an index so that full-file downloads of files that changed can be achieved, as opposed to downloading entire image. If it also chunks the files so that essentially smaller "deltas" can be downloaded then it is a moot point :)

You're right, they are different and not something to compare directly, but this is a feature we have gotten used to and I wasn't sure how to easily achieve a similar concept with RAUC.

ejoerns, 2018-06-11:
I assume this one we can close, right? Feel free to re-open if I was wrong ;)

closed 2018-06-11 by ejoerns