rauc/rauc #276: Slot update fails if partition could not be mounted

mhei, 2018-05-31:
I'm testing rauc on a custom board with a typical A/B rootfs + customerfs setup.

Just for understanding the following trace, the partitioning looks like this:

    label: dos
    label-id: 0x57b0f89d
    device: /dev/mmcblk0
    unit: sectors

    /dev/mmcblk0p1 : start=        2048, size=        6144, type=53
    /dev/mmcblk0p2 : start=        8192, size=     2097152, type=83
    /dev/mmcblk0p3 : start=     2105344, size=     2097152, type=83
    /dev/mmcblk0p4 : start=     4202496, size=     2627584, type=5
    /dev/mmcblk0p5 : start=     4204544, size=     2097152, type=83
    /dev/mmcblk0p6 : start=     6303744, size=      262144, type=83
    /dev/mmcblk0p7 : start=     6567936, size=      262144, type=83

During manufacturing process, p2 is filled with a valid ext4 rootfs, and p5 to p7
are simply formatted as ext4. p3 is left untouched, i.e. usually no valid filesystem
which could be mounted.

When system should be updated the first time, rauc comes into play and it should write the ext4
image contained in the bundle to p3. However, this fails:

    May 30 13:30:44 system rauc[212]: input bundle: /tmp/update-image.raucb
    May 30 13:30:44 system rauc[212]: Active slot bootname: A
    May 30 13:30:44 system rauc[665]: installing /tmp/update-image.raucb: installing
    May 30 13:30:44 system rauc[212]: installing /tmp/update-image.raucb: started
    May 30 13:30:44 system rauc[665]: installing /tmp/update-image.raucb:   0% Installing
    May 30 13:30:44 system rauc[665]: installing /tmp/update-image.raucb:   0% Determining slot states
    May 30 13:30:44 system rauc[665]: installing /tmp/update-image.raucb:  20% Determining slot states done.
    May 30 13:30:44 system rauc[212]: installing /tmp/update-image.raucb: Checking and mounting bundle...
    May 30 13:30:44 system rauc[212]: Reading bundle: /tmp/update-image.raucb
    May 30 13:30:44 system rauc[665]: installing /tmp/update-image.raucb:  20% Checking bundle
    May 30 13:30:44 system rauc[212]: Verifying bundle...
    May 30 13:30:44 system rauc[665]: installing /tmp/update-image.raucb:  20% Verifying signature
    May 30 13:31:33 system rauc[665]: installing /tmp/update-image.raucb:  40% Verifying signature done.
    May 30 13:31:33 system rauc[212]: Mounting bundle '/tmp/update-image.raucb' to '/run/rauc/bundle'
    May 30 13:31:33 system rauc[665]: installing /tmp/update-image.raucb:  40% Checking bundle done.
    May 30 13:31:34 system rauc[665]: installing /tmp/update-image.raucb:  40% Loading manifest file
    May 30 13:31:34 system rauc[665]: installing /tmp/update-image.raucb:  60% Loading manifest file done.
    May 30 13:31:34 system rauc[665]: installing /tmp/update-image.raucb:  60% Determining target install group
    May 30 13:31:34 system rauc[212]: Target Group:
    May 30 13:31:34 system rauc[212]:   customerfs -> customerfs.1
    May 30 13:31:34 system rauc[212]:   rootfs -> rootfs.1
    May 30 13:31:34 system rauc[212]: Starting pre install handler: /usr/lib/rauc/pre-install.sh
    May 30 13:31:34 system rauc[665]: installing /tmp/update-image.raucb:  80% Determining target install group done.
    May 30 13:31:34 system rauc[212]: Using default handler
    May 30 13:31:34 system rauc[212]: Marking target slot as non-bootable...
    May 30 13:31:34 system rauc[665]: installing /tmp/update-image.raucb:  80% Updating slots
    May 30 13:31:34 system rauc[212]: installing /tmp/update-image.raucb: Updating slots...
    May 30 13:37:38 system rauc[212]: Checking image type for slot type: ext4
    May 30 13:37:38 system rauc[212]: Image detected as type: *.ext4
    May 30 13:37:38 system rauc[212]: installing /tmp/update-image.raucb: Checking slot rootfs.1
    May 30 13:37:38 system rauc[212]: mounting slot /dev/mmcblk0p3
    May 30 13:37:38 system rauc[665]: installing /tmp/update-image.raucb:  80% Checking slot rootfs.1
    [ 1238.102459] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
    May 30 13:37:38 system kernel: [ 1238.102459] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
    May 30 13:37:38 system rauc[212]: mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0p3,
    May 30 13:37:38 system rauc[212]:        missing codepage or helper program, or other error
    May 30 13:37:38 system rauc[212]:        In some cases useful info is found in syslog - try
    May 30 13:37:38 system rauc[212]:        dmesg | tail or so.
    May 30 13:37:38 system rauc[212]: Failed to mount slot /dev/mmcblk0p3: failed to mount slot: failed to run mount: Child process exited with code 32
    May 30 13:37:38 system rauc[212]: Slot needs to be updated with /run/rauc/bundle/rootfs.ext4
    May 30 13:37:38 system rauc[212]: Updating /dev/mmcblk0p3 with /run/rauc/bundle/rootfs.ext4
    May 30 13:37:38 system rauc[665]: installing /tmp/update-image.raucb:  85% Checking slot rootfs.1 done.
    May 30 13:37:38 system rauc[212]: installing /tmp/update-image.raucb: Updating slot rootfs.1
    May 30 13:37:38 system rauc[212]: Mounting slot /dev/mmcblk0p3
    [ 1238.441971] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
    May 30 13:37:38 system rauc[665]: installing /tmp/update-image.raucb:  85% Copying image
    May 30 13:37:38 system kernel: [ 1238.441971] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
    May 30 13:37:38 system rauc[212]: mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0p3,
    May 30 13:37:38 system rauc[212]:        missing codepage or helper program, or other error
    May 30 13:37:38 system rauc[212]:        In some cases useful info is found in syslog - try
    May 30 13:37:38 system rauc[212]:        dmesg | tail or so.
    May 30 13:37:39 system rauc[665]: installing /tmp/update-image.raucb:  90% Copying image failed.
    May 30 13:37:39 system rauc[665]: installing /tmp/update-image.raucb: 100% Updating slots failed.
    May 30 13:37:39 system rauc[212]: Handler error: Failed updating slot: failed to mount slot: failed to run mount: Child process exited with code 32
    May 30 13:37:39 system rauc[665]: installing /tmp/update-image.raucb: 100% Installing failed.
    May 30 13:37:39 system rauc[212]: installing /tmp/update-image.raucb: Handler error: Failed updating slot: failed to mount slot: failed to run mount: Child process exited with code 32
    May 30 13:37:39 system rauc[212]: installing /tmp/update-image.raucb: finished
    May 30 13:37:39 system rauc[212]: installing `/tmp/update-image.raucb` failed: 1

I would expect that in such a case rauc "blindly" updates the slot to get it "into shape" again, or do I miss something?
BTW: when I format the slot during manufacturing process, then all works fine.

ejoerns, 2018-06-06:
Hi,

your log truely indicates that the update_handler() call failed.

When I get your logs right, the relevant handler that should have been exectuded (for *.ext4 imgtype to *.ext4 slot type) is the `img_to_fs_handler()`.

This one only mounts when hooks are enabled for this slot. Thus my first assumption is that you have some kind of pre-install hook configured? This one would need a mountable ext4 to successfully execute in current implementation.

Could that be the reason for the issue you encounter?

Nevertheless, I feel we should add more log information about hook execution...

mhei, 2018-06-06:
Yes, you are right. I've specified pre- and post-install hooks for each slot. The pre-install hooks do nothing really important at the moment, other than printing the environment for debugging purpose. The hooks are explicitely exited with "exit 0" to mask any internal error.
I did not look at the code you referenced, but I'm wondering whether the decision about failing at this moment or not could be pushed into the hook in this case, i.e. when slot mount point is empty, the hook could detect this and decide itself whether it requires a mounted fs at this point and return with exit code 1 in this case... (disclaimer: just some quick thoughts - not sure whether this makes sense...)

ejoerns, 2018-06-11:
I would agree that for the pre-install handler it is not necessarily a valid assumption having a mountable file system and that one might want to execute a hook that does not require any pre-existing FS on a slot marked as having an FS.

For the post-install hook this is completely different, of course. After having installed a FS we must be able to mount it, otherwise this is an error.

For now I see two possible solutions for the pre-installer case:

* Always executing the hook without mounting the FS, but this might limit use cases and potentially breaks compatibility
* Executing the hook either mounted or unmounted (as you suggested)

The drawback of the latter solution I see is reduced predictability and shifted responsibility, but it might be a proper solution anyway. @jluebbe what do you think about that?

mhei, 2019-09-18:
Since I just stumpled over this one year old bug again, I discussed it with management and we tend to prepare a PR to fix it. However, I would need to know about the direction: I still vote for executing the hook unconditionally. Passing the mount point in environment, or not passing it, could allow the hook to react in both cases.
Please advice howto proceed @jluebbe and @ejoerns. Thanks!