rauc/rauc #281: Support for extended file attributes (XATTR) when writing a slot from a compressed tar archive?

krobbins, 2018-07-03:
My embedded distribution integrates SELinux, which depends on extended file attributes for the file security contexts. The root file system is properly labeled upon construction of the compressed tar archive.

If I manually extract the tar archive to an Ext4 file system mounted with XATTR support, the SELinux labels are correct and I can boot into the new system without relabeling on the first boot.

It seems that the RAUC defaults do not mount the Ext4 file system with XATTR support, because when I install a bundle with RAUC and then reboot into the new system the root file system is relabeled, which forces a reboot. This forced reboot defeats the boot loader (EFI) logic and the system reboots into the previous system.

krobbins, 2018-07-03:
I got to thinking that the XATTR security contexts might also be missing in the installed system because of the options to the tar command used in the function untar_image(). My distribution includes GNU tar 1.30. From the tar man page it seems that if the archive includes ACLs, XATTR, and SELinux contexts, then no special options are needed when extracting the archive.

But, when I untar the compressed archive by hand, I always specify --numeric-owner, which ensures the intended user/group IDs are written to disk. It may be that without this option some of the files in the archive are extracted without the intended user/group IDs from the archive and have to be relabeled on the first boot.

krobbins, 2018-07-03:
I've worked around this issue for the time being in the post-install slot hook, where I chroot into the new system and run "fixfiles -f relabel" to ensure the correct SELinux security contexts exist:

                # Setup a chroot to relabel the root file system
                for d in dev sys proc; do mount --bind /$d ${RAUC_SLOT_MOUNT_POINT}/$d; done
                chroot ${RAUC_SLOT_MOUNT_POINT}/ fixfiles -f relabel
                umount ${RAUC_SLOT_MOUNT_POINT}/{dev,sys,proc}

Now, when rebooted the machine successfully boots into the newly installed system, because no relabeling followed by a reboot is needed.

ejoerns, 2018-07-05:
Note that with #284 RAUC now uses the `--numeric-owner` option for tar extraction which should fix at least one part of this issue.

krobbins, 2018-07-05:
My default mount options are currently (from `/proc/mounts` on my embedded target):

`/dev/root / ext4 rw,seclabel,relatime,data=ordered`

I would certainly welcome the addition of `mount-options` to the `system.conf`.

Also, thanks for the tip to use the pre-install slot hook to remount the file system. I was definitely in brute-force mode a two days ago when I broke out the chroot-hammer.

ejoerns, 2018-08-03:
@krobbins Would you like to make a patch for this? :)

PR #338 (2018-10-10): Add extra-mount-opts slot config key
Allow passing custom slot config options to RAUC's slot mount calls.

This fixes #281.
closed 2018-10-10 by jluebbe