rauc/rauc #288: install --keyring does not work with rauc.service

ilkondr, 2018-07-12:
```
root@board:/tmp# rauc --keyring=/etc/rauc/keyring.pem install bundle.raucb
rauc-Message: installing /tmp/bundle.raucb: installing
rauc-Message: installing /tmp/bundle.raucb:   0% Installing
rauc-Message: installing /tmp/bundle.raucb:   0% Determining slot states
rauc-Message: installing /tmp/bundle.raucb:  20% Determining slot states done.
rauc-Message: installing /tmp/bundle.raucb:  20% Checking bundle
rauc-Message: installing /tmp/bundle.raucb:  40% Checking bundle failed.
rauc-Message: installing /tmp/bundle.raucb: 100% Installing failed.
rauc-Message: installing /tmp/bundle.raucb: LastError: No keyring file provided
Installing /tmp/bundle.raucb failed
```

However `rauc info --keyring=$keyring_path bundle.raucb` works well.
`install` works for me only if keyring path provided with default conf file. So that means `rauc install -c $config_path bundle.raucb` also does not work even if keyring path is provided with custom config file.

jluebbe, 2018-07-13:
You need to set the keyring to use in the config file used by the service. The expectation is that you only need one config file on any given system.
When used via D-Bus, the client should reject the these options as they don't have an effect on the service, to avoid confusing the users.

ejoerns, 2018-07-27:
@ilkondr Did that answer your question?

I've opened #292 to face the issue @jluebbe mentioned

ejoerns, 2018-08-21:
As there was no response and the issue behind it is addressed by another PR, I'll close that for now. If there are questions remaining feel free to reopen.

closed 2018-08-21 by ejoerns