rauc/rauc #301: Question/Support

hpatriarche, 2018-08-09:
Hi guys,

In evaluation software update tool, I am testing RAUC which look pretty nice.
I am trying to use hook install concept as I want to use a specific installation script. But it does not seem to work.

Here is my 
```
# cat /etc/rauc/system.conf 
[system]
compatible=BMS-Box
bootloader=uboot

[keyring]
path=ws.cert.pem

[slot.rootfs.0]
device=/dev/mtd8
bootname=A

[slot.rootfs.1]
device=/dev/mtd9
bootname=B

Manifest looks like:
[update]
compatible=BMS-Box
version=2018.08-1

[hooks]
filename=hook

[image.rootfs]
sha256=5b594096285451ce431cbc0a1462a27751f1b489029a4669a026f17aa724a423
size=34537472
filename=rootfs.ubifs
hooks=install;
```

and hook script:
```
#!/bin/sh

echo "Hooks ..." > install.log
echo "param 1= $1" >> install.log

case "$1" in
        slot-install)
                # only rootfs needs to be handled
                echo "Class: $RAUC_SLOT_CLASS" >>  install.log
				echo "Device: $RAUC_SLOT_DEVICE" >>  install.log
                ;;
        *)
				echo "Other case exit 1" >> install.log
                exit 1
                ;;
esac

exit 0
```

But when I execute it on target:

> # rauc install -d update-2018.08-1.raucb 
> Domains: 'rauc'
> (rauc:735): rauc-DEBUG: 15:47:54.613: install started
> (rauc:735): rauc-DEBUG: 15:47:54.615: input bundle: /root/update-2018.08-1.raucb
> (rauc:735): rauc-DEBUG: 15:47:54.674: Trying to contact rauc service
> rauc-Message: 15:47:54.690: installing /root/update-2018.08-1.raucb: installing
> rauc-Message: 15:47:54.702: installing /root/update-2018.08-1.raucb:   0% Installing
> rauc-Message: 15:47:54.705: installing /root/update-2018.08-1.raucb:   0% Determining slot states
> rauc-Message: 15:47:54.712: installing /root/update-2018.08-1.raucb:  20% Determining slot states done.
> rauc-Message: 15:47:54.720: installing /root/update-2018.08-1.raucb:  20% Checking bundle
> rauc-Message: 15:47:54.733: installing /root/update-2018.08-1.raucb:  20% Verifying signature
> rauc-Message: 15:47:56.316: installing /root/update-2018.08-1.raucb:  40% Verifying signature done.
> rauc-Message: 15:47:56.330: installing /root/update-2018.08-1.raucb:  40% Checking bundle done.
> rauc-Message: 15:47:56.392: installing /root/update-2018.08-1.raucb:  40% Loading manifest file
> rauc-Message: 15:47:56.404: installing /root/update-2018.08-1.raucb:  60% Loading manifest file done.
> rauc-Message: 15:47:56.412: installing /root/update-2018.08-1.raucb:  60% Determining target install group
> rauc-Message: 15:47:56.432: installing /root/update-2018.08-1.raucb:  80% Determining target install group done.
> rauc-Message: 15:47:56.600: installing /root/update-2018.08-1.raucb:  80% Updating slots
> rauc-Message: 15:48:04.204: installing /root/update-2018.08-1.raucb:  80% Checking slot rootfs.1
> rauc-Message: 15:48:04.211: installing /root/update-2018.08-1.raucb:  90% Checking slot rootfs.1 done.
> rauc-Message: 15:48:04.218: installing /root/update-2018.08-1.raucb:  90% Copying image skipped
> rauc-Message: 15:48:04.232: installing /root/update-2018.08-1.raucb: 100% Copying image skipped done.
> [ 6654.060947] UBIFS error (pid: 746): cannot open "/dev/mtdblock9", error -22
> rauc-Message: 15:48:04.628: installing /root/update-2018.08-1.raucb: 100% Updating slots done.
> rauc-Message: 15:48:04.961: installing /root/update-2018.08-1.raucb: 100% Installing done.
> Installing `/root/update-2018.08-1.raucb` succeeded

I wonder why I have no file install.log  and why I got message:
 6654.060947] UBIFS error (pid: 746): cannot open "/dev/mtdblock9", error -22

Can you explain what I am doing wrong ?

Thanks for you help.

Hervé.

hpatriarche, 2018-08-09:
It works if I specify 
```
[image.rootfs]
filename=rootfs.img
size=...
sha256=...
hooks=pre-install
```
but not with ```

[image.rootfs]
filename=rootfs.img
size=...
sha256=...
hooks=install
```

hpatriarche, 2018-08-10:
some progress:
ubisfs message :

> UBIFS error (pid: 746): cannot open "/dev/mtdblock9", error -22

is due to type defined into system.conf; it looks when specify hooks=install we should not specify any type.

Then regarding install hook, it works after first booting the kernel and at first 
`rauc install <bundle>`

but if I rerun the same command, I have message output but hook is not called.
I try to change some uboot variable BOOT_ORDER, but it is not helping.

So now question is:
what should be done to rerun 
`rauc install <bundle>`
twice ?

Thanks.

hpatriarche, 2018-08-10:
Finally got it !
In order to rerun:
`rauc install <bundle>`
  333 root     {rauc-service.sh} /bin/sh /usr/libexec/rauc-service.sh
  334 root     /usr/bin/rauc service
must be killed

ejoerns, 2018-08-10:
Great you already solved it yourself!
Yes, for changes on configuration to take effect, the service must be restarted. The command-line client does not parse configuration during installation. Glad it worked than as you expected!

hpatriarche, 2018-08-10:
Yep, maybe the doc should be updated ?
Also a minor issue, I still not able to handle:
when I do some echo in hook, I got nothing on output. Only if I output something on a file, I could see it. Do you have any idea ?

hpatriarche, 2018-08-10:
Humm, I updated system.conf with:

> [slot.rootfs.1]
> force-install-same=true
> device=/dev/mtd9
> bootname=B
>

But same behavior, I mean when I run
killall rauc
rauc -d install /root/update-2018.08-3.raucb  => here hook is called
rauc -d install /root/update-2018.08-3.raucb => second times, hook is not called

Also if I added

> [slot.rootfs.1]
> device=/dev/mtd9
> bootname=B
>type=ubifs

Then I have again message:

> [ 5931.910332] UBIFS error (pid: 1012): cannot open "/dev/mtd9", error -22

I am using V0.4

ejoerns, 2018-08-10:
@hpatriarche seem GitHub did not show me your comment when I wrote mine.

About the doc update: I am note sure yet where to note this. From the basic design (client/server) this should be clear but I would agree that one might not be aware of this when being focused on testing integration. Thus maybe it is worth adding a line or note in system configuration chapter.

The echo is printed nowhere? Note that the client only output status information it receives via D-Bus. For reading the actual log you must have a look at what the service emits. i.e. when using systemd: `journalctl -u rauc`.

Now about your recent commit, which I fortunately just saw: I have no idea :) Seems I have to test this and then come back to you with my findings. ;)

hpatriarche, 2018-08-10:
@ejoerns many thanks for your help and feedback.
For now I am able to manage what system should do, so it under control

commit d17b1ea8 (2018-10-10): docs/reference: add note on how to change system.conf on running target

Fixes #301.
closed 2018-10-10 by jluebbe