rauc/rauc #303: Error on "make"

ckilimci, 2018-08-10:
Hi all,

I am able to run autogen.sh and configure successfully but I am getting an error while "make".

Here is the output of make:

$ make
  GEN      rauc-installer-generated.c
make --no-print-directory all-am
  CC       src/librauc_la-bootchooser.lo
  CC       src/librauc_la-bundle.lo
  CC       src/librauc_la-checksum.lo
  CC       src/librauc_la-config_file.lo
  CC       src/librauc_la-context.lo
  CC       src/librauc_la-emmc.lo
  CC       src/librauc_la-install.lo
  CC       src/librauc_la-manifest.lo
  CC       src/librauc_la-mark.lo
  CC       src/librauc_la-mount.lo
  CC       src/librauc_la-service.lo
  CC       src/librauc_la-signature.lo
  CC       src/librauc_la-utils.lo
  CC       src/librauc_la-update_handler.lo
  CC       src/librauc_la-network.lo
  CC       librauc_la-rauc-installer-generated.lo
  CCLD     librauc.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CC       src/rauc-main.o
  CCLD     rauc
/usr/bin/ld: warning: libssl.so.1.0.2, needed by /usr/lib/x86_64-linux-gnu/libcurl.so, may conflict with libssl.so.1.1
/usr/bin/ld: warning: libcrypto.so.1.0.2, needed by /usr/lib/x86_64-linux-gnu/libcurl.so, may conflict with libcrypto.so.1.1
collect2: error: ld returned 1 exit status
Makefile:1041: recipe for target 'rauc' failed
make[1]: *** [rauc] Error 1
Makefile:854: recipe for target 'all' failed
make: *** [all] Error 2

$uname -a
Linux deb 4.9.0-7-amd64 #1 SMP Debian 4.9.110-3+deb9u1 (2018-08-03) x86_64 GNU/Linux

$$gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.3.0-18+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)

I am sending my confugure and Makefile with txt extension.
[configure.txt](https://github.com/rauc/rauc/files/2278239/configure.txt)
[Makefile.txt](https://github.com/rauc/rauc/files/2278240/Makefile.txt)

Any idea helps a lot. Thank you for your time.
Caglar

ejoerns, 2018-08-10:
According to the error message

```
/usr/bin/ld: warning: libssl.so.1.0.2, needed by /usr/lib/x86_64-linux-gnu/libcurl.so, may conflict with libssl.so.1.1
/usr/bin/ld: warning: libcrypto.so.1.0.2, needed by /usr/lib/x86_64-linux-gnu/libcurl.so, may conflict with libcrypto.so.1.1
```

libcurl is build against OpenSSL 1.0.2 while the detected and used version of OpenSSL on your system is 1.1. 

On Debian / Ubuntu you can fix this by installing header for 1.0.x instead (which will conflict with `libssl-dev` and requires us to remove it) the following way:

```
sudo apt-get purge libssl-dev
sudo apt-get install libssl1.0-dev
```

ckilimci, 2018-08-12:
Perfect! That worked. Thank you.

closed 2018-08-12 by ckilimci