rauc/rauc #320: Allow setting casync workdir location

elboulangero, 2018-08-30:
By default, casync working directories are created in `/var/tmp`. However, this can be modified by setting the TMPDIR environment variable.

CASync working directories can get quite big, in size and number of inodes. Hence, for some use-case it might be preferable to ensure that all this work is done in RAM rather than on-disk. This can be achieved simply by setting TMPDIR to a location that is backed by RAM. For example, one can set it to `/tmp`, assuming that /tmp is mounted as tmpfs.

The upside of using tmpfs for casync updates:
- no housekeeping (if there's a crash, we're sure there's no leftovers after reboot)
- for a tmpfs, it's possible to modify the maximum number of inodes and remount at any time. It's very convenient, and it can be done just before starting an update, to ensure that the greedy casync won't run out of inodes.

So, this could be implemented easily in RAUC, by adding a `tmpdir` setting to the casync section of the configuration file.

Would you accept patches for this feature?

Thanks!

ejoerns, 2018-08-30:
In general, we take all patches that add a reasonable enhancement to RAUC or solve issues that users have with it. They must not break backwards compatibility and they must apply to our quality standards.

Thus, yes I'm pretty sure that we would accept a patch for the use case you described as I assume that changing the tempdir might be a requirement for others too and behavior is not changed if one does not use the `tmpdir` setting.

jluebbe, 2018-08-30:
@elboulangero It should be possible to just use the TMPDIR environment variable from outside of rauc. For which use-case do you prefer to set it via the rauc config file?

elboulangero, 2018-08-30:
@jluebbe You're right, I just didn't think about it TBH. That's even more simple.

Ok I'll close this issue then, thanks!

closed 2018-08-30 by elboulangero

jluebbe, 2018-08-30:
OK. :) If this doesn't actually work, that would be a bug.