rauc/rauc #321: Question/Casync - Consider seeding output device?

elboulangero, 2018-08-31:
If I'm not mistaken, at the moment RAUC only uses the **active slot** for casync-seeding:

https://github.com/rauc/rauc/blob/dfb3251db5275065b88f940e86f41a85d6a730b7/src/update_handler.c#L183

https://github.com/rauc/rauc/blob/dfb3251db5275065b88f940e86f41a85d6a730b7/src/update_handler.c#L241

Did you consider the possibility to seed the **target slot** instead?

The benefit I see is for the case where an update is interrupted. If we seed the active slot, then anything that could have been written won't be re-used, we will have to restart the update from zero, and re-download every chunk. While if we seed the target slot, we kind of recover, and every chunk that was written is treated as a seed, and doesn't need to be downloaded again.

If you consider a theoretical situation where download takes a lot of time (lot of data, slow connection), and is likely to be interrupted, then it's interesting to know that each time we will retry an update, we have less and less data to download, until we finally succeed.

There's one downside I see though: the very first update is an issue, as the target slot is probably empty, and it means casync would have to download everything. So this first-update case needs to be handled one way or another.

I'm not really facing these issues, I'm still in the testing stage, so I open this issue just for discussion, I'd like to know what you think about it.

Thanks!

ejoerns, 2018-10-03:
@elboulangero sorry for letting you wait a little longer with my response.

We also discussed if using the active or the target slot is more appropriate, ending up in using the active one as you see in current implementation. But, as you noted, there are also good reasons for using the target device instead, or even additionally.

The default behavior of casync is actually to use the target device for seeding and at the moment we explicitly deactivate this with the `--seed-output=no` parameter.

In the end all that we do with casync is always a balance of several trade-offs. Seeding always increases update time on the target while potentially lowering download volume. It is very use-case specific at which adjusting screws you have to turn to gain the best results for your use case.

To make a long story short, I would not see it as the default use case, but see that there might be reasons for using output as seed, too. Thus if that is required, you might like to add an option to the `[casync]` system.conf section that enables output seeding conditionally.

elboulangero, 2018-10-03:
@ejoerns Thanks for your feedback. I was also thinking about an option in `system.conf`, I wouldn't make it the default.

I'm still unsure if I will really need that, and also not sure about how to best handle the first update, in case the target slot is empty.

But it's good to hear that you would be interested in that feature. If ever I go there, I'll make sure to submit a patch.

ejoerns, 2019-03-26:
@elboulangero Have you already become clearer about this? ;)

Any idea of how we should proceed here? Has anyone else a similar use case or is interested to test such an option?

elboulangero, 2019-04-02:
@ejoerns I didn't do any progress on that, and I don't think I'll be doing anything in the coming weeks either. This is stalled for now.