rauc/rauc #325: install hook and other bundled files are not taken into account in checksum

Villemoes, 2018-09-10:
When implementing a custom install hook, changing the logic in that script should result in a "checksum mismatch", even if a bundle with the same image file has already been installed. More generally, if any custom hooks are in play, any file in the bundle can affect the result, so should somehow be taken into account before deciding that the update should/may be skipped.

I realize it's not that easy to implement. Thinking out loud: Perhaps the manifest would have to have a per-slot "extra_sha256", and a corresponding "extra_files" indicating which files in the bundle should go into computing that (computed as "cat $extra_files | sha256sum" or something). Normally, extra_sha256 would be just sha256 of the empty string, and that's what would be used for comparison when an existing slot status file does not have an extra_sha256.

Perhaps a simpler thing to implement is supporting force-install-same on a per-manifest basis.

Villemoes, 2018-09-11:
Actually, there's no need for an extra_sha256, only an extra_files (or whatever name is appropriate) that would also be included in the existing sha256. And looking closer, it actually seems rather straight-forward to implement (just have update_checksum and verify_checksum take a list of files instead of just one), but I'll wait a bit more for people to comment before making a POC patch.

ejoerns, 2018-09-14:
Well, first of all I agree with you that when using hooks the relation between the sha and the actual content that will be installed is not that clear. As you pointed out

1) The hook script might change logic
2) The image(s) that will be installed may change
3) Other scripts executed by the hook script may change

Taking care of the first one seems straight forward as we only use a single (base) hook script/binary. Thus each slot that has a hook defined will take the hook script into account for calculating its sha.

But, beside 'slot hooks' there are also 'install hooks' that are global and can affect the entire installation. I am not sure how to handle these, yet. At the moment we only have the 'install-check' hook that would only have impact on *if* something will be installed or not. But thinking of per-install post-install hooks I am not fully sure if we would have to take them into account or can argue that they only change what will happen before/after installation and will be executed anyway...

For images there is no drawback when using only a single one, the one we reflect in the `filename` key.
But as you pointed out, multiple files/images may be used by a single slot hook where you can do whatever you like. Maybe you have a bootloader + bitstream that you want to write into the same slot, or have an additional script being called by the hook script that performs data migration for you slot.

Adding these as `extra_files` to the `[image]` section in the manifest they belong to and allow tracking them sounds reasonable for me.

The remaining point to discuss is if one would use a separate sha for this or include this in the existing sha.
The question is if there is an advantage on having the images sha separately and if there are backwards compatibility issues.
But at a first glance I would also argue for adding this to the existing sha to keep it simple. The worst thing that may happen when migrating to the new hashing is that the system re-installs an identical bundle which seems pretty undramatic.

I do not see any great advantage in using `force-install-same` on a per manifest base, yet. Except that one would have to write only a single key instead one per-slot that should be forced.

A more interesting thing would be to allow forcing installation on a per-installation base, independent from the bundle. E.g. if you know something in the system is strange, maybe because it was badly modified at some point after the installation.
But that is currently blocked by the D-Bus API that does not support further arguments  for the install method and that we cannot simply extend.

@jluebbe what is your opinion on this entire topic?

Villemoes, 2018-10-12:
So I went ahead and tried to refactor the checksumming code to make it easy to hook up some "also checksum these files", see #348 .