rauc/rauc #329: Chat room : kick from it ...

hpatriarche, 2018-09-17:
Hi,
Currently no way to join the RAUC room chat:

> You have been kicked from #rauc by @appservice-irc:matrix.org.
> Reason: IRC error on #rauc: err_needreggednick

Any clue ?

Thanks, Herve.

ejoerns, 2018-09-17:
@hpatriarche freenode had some trouble with spam which required us to set the channel access modes to be more strict.
This also seems to prevent access from matrix currently. Direct access with a registered freenode nick will work instead.
Btw. the adjusted permissions are also the reasons why one does not see pull-request and issue messages no the #rauc IRC currently

hpatriarche, 2018-09-17:
Thanks Joerns, Now I am in the room.

ejoerns, 2018-12-10:
In the meantime I've changed permissions back and also the GitHub irc client is allowed to push updates again. Thus we can close this.

closed 2018-12-10 by ejoerns