rauc/rauc #330: How to get log messages output

hpatriarche, 2018-09-17:
Hi,
I am struggling to get more output message.
I would like to get message from g_message statement. But neither in console output or in /var/log/messages I get any of those log.
What the way to get those log ?

Thanks, Herve.

ejoerns, 2018-09-17:
@hpatriarche, depends on how you start your service.

If you use the default systemd way, the log will go into systemd journal and you can access it as for every unit using

    journalctl -u rauc

hpatriarche, 2018-09-17:
Humm, I am with sysvinit
So I do not have journalctl tool

ejoerns, 2018-09-17:
It really depends on how you start the RAUC service. RAUC simply logs to stdout.
Everything else is up to your service script.

So, how do you start the service? With init script during system startup / D-Bus activation?..
How does it look like?

In general, with sysvinit scripts, one uses something like

    rauc service > /var/log/rauc.log 2>&1

or the `logger` tool.

And, why not using systemd?

ejoerns, 2018-10-03:
@hpatriarche did that solve your problems?

hpatriarche, 2018-10-03:
@ejoerns yes, I fixed it. Thanks.

ejoerns, 2018-10-03:
@hpatriarche Great! Thanks for your answer, thus I can safely close this thread.

closed 2018-10-03 by ejoerns