rauc/rauc #334: Calling rauc install from udev fail

AmarOk1412, 2018-10-04:
Hi!

I currently use `rauc` (0.1.1) to update some bundle on a custom distrib made with yocto. The bundle comes from a USB device. So, I add a udev rule to get my bundles as soon as the key is inserted on the device.

My hook contains (version A):

```bash
echo ${PATH} #/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin
echo `fw_printenv slot` #slot=A
rauc_output=`rauc install ${bundle}`
echo ${PATH} #/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin
```

But with this version, it fails with `starting fw_setenv failed: Failed to execute child process "fw_setenv" (No such file or directory)` if launched via udev (If I launch manually the script it works).

If I use this version:

```bash
echo ${PATH} #/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin
echo `fw_printenv slot` #slot=A
rauc_output=`PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin rauc install  ${bundle}`
echo ${PATH} #/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin
```

It works (both manually and with udev)

I don't really know what the heck is happening here. The PATH is ok before and after the install, but... not when calling rauc via udev. Moreover, `fw_setenv` is correctly found if I call it just before `rauc`.

The code to call `fw_setenv` in rauc is (4b6453e8678a48aa793c41871e3bfcfdbafce881):

```c
	sub = g_subprocess_new(G_SUBPROCESS_FLAGS_NONE, &error, UBOOT_FWSETENV_NAME,
			       key, value, NULL);
	if (!sub) {
		g_warning("starting fw_setenv failed: %s", error->message);
		g_clear_error(&error);
		goto out;
	}
```

`g_subprocess_new` is searching in `PATH`. So this should be ok.

Any clue?

ejoerns, 2018-10-04:
Hi @AmarOk1412 

I assume you compiled RAUC with service enabled (default)? Then it is actually the *service* that must have the right PATH set, not the client (the CLI).

If you use D-Bus activation and do not use systemd, you might run into the issue fixed in https://github.com/rauc/rauc/commit/dcd184d7c89c78c5ed957a02169e836a236d4ece.

Anyway, I am not aware what makes your second example work as I would assume this to equal you first one..

AmarOk1412, 2018-10-04:
No, I don't use the service for another issue.

For my issue I don't think it's very related to `rauc` but more `udev`. I also opened a ticket on SO because I don't have any clue for now. Anyway, if I have some time to dig, it's not very critical.

ejoerns, 2018-10-04:
Sorry but I didn't get what you mean with your first sentence.

If I get it right, you start RAUC with `RUN+=/path/to/script.sh` from your udev rule? And you compiled RAUC with `--disable-service`?

I saw the post on SO, but I don't assume you will get more targeted help there...

jluebbe, 2018-10-05:
> No, I don't use the service for another issue.
> 
> For my issue I don't think it's very related to `rauc` but more `udev`. I also opened a ticket on SO because I don't have any clue for now. Anyway, if I have some time to dig, it's not very critical.

udev kills your process. From udev(7):
```
           This can only be used for very short-running foreground tasks. Running an event
           process for a long period of time may block all further events for this or a dependent
           device.

           Starting daemons or other long-running processes is not appropriate for udev; the
           forked processes, detached or not, will be unconditionally killed after the event
           handling has finished.
```

AmarOk1412, 2018-10-05:
@ejoerns yes for both questions. And yup I don't think I will have an answer on SO.

@jluebbe not the issue here, the timeout is very huge and nothing to do with `fw_printenv`

ejoerns, 2018-12-10:
@AmarOk1412 does this still happen or have you been able to solve it?

AmarOk1412, 2018-12-10:
it still happen with udev, but I don't care we use a cron to do the job now. I close this.

closed 2018-12-10 by AmarOk1412