rauc/rauc #337: Override keyring certificate file

stobajas, 2018-10-09:
Hello,
I'm using RAUC v0.1.1 with Yocto Pyro 2.3/meta-rauc.
As rauc will be called with a udev script (thanks to @AmarOk1412 ), only the udev script knows the path of the certificate file.
So I want rauc to be called as:
` rauc install --cert=/path/to/certificate <rauc_bundle> `
So, I have done the following patch on the file src/bundle.c based on the tag v0.1.1:
```c
diff --git a/src/bundle.c b/src/bundle.c
index fd9b80f..85de36c 100644
--- a/src/bundle.c
+++ b/src/bundle.c
@@ -260,11 +260,17 @@ gboolean check_bundle(const gchar *bundlename, gsize *size, gboolean verify, GEr
 
 	r_context_begin_step("check_bundle", "Checking bundle", verify);
 
-	if (!r_context()->config->keyring_path) {
+	if (!r_context()->config->keyring_path && !r_context()->certpath) {
 		g_set_error(error, G_FILE_ERROR, G_FILE_ERROR_EXIST, "No keyring file provided");
 		goto out;
 	}
 
+	// if keyring_path does not exist but a certpath has been specified
+	// override the keyring_path with the certpath
+	if (!r_context()->config->keyring_path && r_context()->certpath) {
+		r_context()->config->keyring_path = g_strdup(r_context()->certpath);
+	}
+
 	g_message("Reading bundle: %s", bundlename);
 
 	bundlefile = g_file_new_for_path(bundlename);
```
I have some questions:
1. Is it wanted feature?
2. Is there another way to link an external certificate (other than doing a symlink)?
3. I have not tested if this feature is implemented in the master branch. If this feature is wanted, i can easily do a pull request (with the previous patch).

Thanks for your comments !

ejoerns, 2018-10-09:
Hi @stobajas,

for verifying the signature of a bundle, you usually do not use a certificate but a keyring instead.
There is only one special case where these (the certificate and the keyring) equal: when not using real PKI but self-signed certificates instaed. I assume that this is your use-case, right?

In upstream RAUC, there is 1176df7eec9eca4c209a33a2e7bcf9e26abef4de that is in since v0.2. It adds a `--keyring` option that does exactly what you need.

I would advice you to use a more recent RAUC version, anyway. 0.1.1 is quite obsolete and I would always recommend starting with state-of-the-art software when developing a platform.

I also emphasize what @jluebbe already mentioned in the other thread: udev's `RUN` is really not made for simulating a service manager. You could use a combination of udev+systemd mechanis for this.

stobajas, 2018-10-10:
Hi @ejoerns, 
Thanks for your response.
Yes you are right, for now, we don't have a real PKI as our board is always offline and we don't want to embed a PKI in the board as we still have sysV init system.
I have planned later to bump Rauc to v0.4 to enjoy the keyring and casync features ;).

Thanks for the advices, you are doing a great work !

closed 2018-10-10 by stobajas