rauc/rauc #343: Segmentation fault with tagged version

jfrantisek, 2018-10-10:
Hello,
I'm trying to setup rauc on ubuntu 18.04 x86_64 system.
Compiling and running from master is successful. But I get segmentation fault after `rauc status` command when I try to use v0.4 version (https://github.com/rauc/rauc/archive/v0.4.zip). Same problem is on v0.3.
I follow these commands:
```
apt update && apt install automake libtool libglib2.0-dev libcurl4-openssl-dev libssl-dev libjson-glib-dev unzip -y
mkdir /etc/rauc  
cp *.crt *.conf /etc/rauc
wget https://github.com/rauc/rauc/archive/v0.4.zip
unzip v0.4.zip
cd rauc-v0.4
./autogen.sh
./configure
make
make install
rauc status
```
Do you know where can be a problem?

ejoerns, 2018-10-10:
@jfrantisek could you provide a log of the call that resulted in the segfault?

If you call it with `-d` (debug) option it also prints some additional information potentially worth for encircling the issue.

Just from the steps above, which are very unspecific, it is really hard to guess what might have gone wrong.

jfrantisek, 2018-10-10:
```
rauc -d status
Domains: 'rauc'
(rauc:1056): rauc-DEBUG: 18:12:47.571: status start
(rauc:1056): rauc-DEBUG: 18:12:47.572: Resolved bootname /dev/disk/by-uuid/7cbbe509-ec9f-4a0b-bbf3-c549bfbbc6ed to /dev/sda3
(rauc:1056): rauc-DEBUG: 18:12:47.573: Found external mountpoint for slot rootfs.0 at /
(rauc:1056): rauc-DEBUG: 18:12:47.573: Found external mountpoint for slot efi.0 at /boot/efi
(rauc:1056): rauc-DEBUG: 18:12:47.574: Found booted slot: rootfs.0 on /dev/sda3
```
What do you mean by unspecific? Specific way is use yocto, ptxdist etc?

ejoerns, 2018-10-10:
With unspecific i meant that only having the information that it segfaults makes it really hard to trace down where in the code the segfault originates from ;)

As there is no segfault message in your log output, I assume this happened right after the last line you posted?
Do you have the option to start rauc with gdb and dump a stack trace?

``` sh
$ gdb --args rauc status
(gdb) run
[...] segfault
(gdb) bt
(gdb) quit
```

Could you verify if segfault still happens with current master branch?

ejoerns, 2018-10-10:
Probably you run into segfault fixed by https://github.com/rauc/rauc/commit/6b0687e8915947520ee01a4f3ed3699fb42b0539 ?

jfrantisek, 2018-10-11:
[6b0687e](https://github.com/rauc/rauc/commit/6b0687e8915947520ee01a4f3ed3699fb42b0539) solve the problem.
Thank you for help!

closed 2018-10-11 by jfrantisek