rauc/rauc #347: Update of Certificate Revocation List

embeddedrocks, 2018-10-12:
Hi guys,

We are using RAUC to update our system and it works seamlessly.

But there is a question we came across to completely close the 'security' loop. This is regarding the general process of using the certificate revocation list.

Our understanding is that the certificate contains a link to the CRL that needs to be downloaded. The content of that CRL is then checked against the certificate if it is still valid.
But RAUC does not support the download of the CRL (checked your source code as well). Is that correct?

In your test CA's you embed the CRL into the certificate. So this would allow for an offline check. But this does not help in general.
If an adversary get's hold of the private key, he/she can create their own RAUC bundles. So we would revoke the certificate and create a new one which embedds the latest CRL.
But as the revoked certificate still embedds the old CRL, the RAUC bundle will still be accepted and installed.

How do we overcome the problem? Or do we miss something fundamentally? Is there a cache openssl uses, so the CRL can be downloaded by a third party application?

embeddedrocks, 2018-10-15:
@jluebbe Many thanks for your quick response!

Just to make sure I got it right: There is currently no support for certificate revocation. This would need to be implemented by RAUC?!

As of the general processing of CRL/OCSP responses, it is not possible to have a offline revocation? In case of deploying a new update which does contain the latest revocations and therefore dismiss older updates as the certificate is already revoked.

Many thanks

jluebbe, 2018-10-18:
> @jluebbe Many thanks for your quick response!
> 
> Just to make sure I got it right: There is currently no support for certificate revocation. This would need to be implemented by RAUC?!

No. If you have a CRL in your keyring (similar to how the test CA handles it), RAUC will check that CRL. RAUC doesn't handle _updating_ the CRL.

> As of the general processing of CRL/OCSP responses, it is not possible to have a offline revocation? In case of deploying a new update which does contain the latest revocations and therefore dismiss older updates as the certificate is already revoked.

Yes, this is possible (by including a new CRL in the keyring in the new image).

embeddedrocks, 2018-10-19:
> No. If you have a CRL in your keyring (similar to how the test CA handles it), RAUC will check that CRL. RAUC doesn't handle updating the CRL.

Clear.

> As of the general processing of CRL/OCSP responses, it is not possible to have a offline revocation? In case of deploying a new update which does contain the latest revocations and therefore dismiss [...]

Understood, but it still does not solve the revocation issue. Example:
We are using (key1, cert1) to sign image1. Now cert1 gets compromised as an adversary gets hold of key1. Therefore cert1 is revoced and CRL is updated. Then (key2, cert2 [including CRL]) are created and will be used to sign image2. Now the adversary creates image3 with compromised key1. This image does not contain a CRL. So this image will be accepted although cert1 is revoced.

SilentHunter124, 2021-02-28:
Just a question to this topic: When downloading CRL is done once a day by a cron job and put in the crl directory. Does rauc validate it correctly? I mean especially the usage of an indirect CRL (signed by another root ca)? And of course is it rejected (not soft fail) when a crl is expired?

ejoerns, 2021-02-28:
Ok, if we pick this up, we should add some update and answer @embeddedrocks first ;)

> > No. If you have a CRL in your keyring (similar to how the test CA handles it), RAUC will check that CRL. RAUC doesn't handle updating the CRL.
> 
> Clear.
> 
> > As of the general processing of CRL/OCSP responses, it is not possible to have a offline revocation? In case of deploying a new update which does contain the latest revocations and therefore dismiss older updates as the certificate is already revoked.
> > > Yes, this is possible (by including a new CRL in the keyring in the new image).
> 
> Understood, but it still does not solve the revocation issue. Example:
> We are using (key1, cert1) to sign image1. Now cert1 gets compromised as an adversary gets hold of key1. Therefore cert1 is revoked and CRL is updated. Then (key2, cert2 [including CRL]) are created and will be used to sign image2. Now the adversary creates image3 with compromised key1. This image does not contain a CRL. So this image will be accepted although cert1 is revoced.

If I got your scenario right, then image3 would be rejected because it is signed with key1 which is in the current rootfs' CRL. Before installation, it does not matter if the image to be installed contains a CRL or not. It's the CRL of the currently running rootfs that matters.

Now in general: What was noted back in 2018 about CRLs was not true anyway, as against our assumption, the CRL checking needs to be enabled manually in OpenSSL. This was then introduced with #561 and needs to be activated expliciticly via `check-crl=true` in system.conf (because of backwards-compatibility reasons). However, since this PR RAUC will also warn if it sees a CRL but `check-crl` is not set.

@SilentHunter124 This brings us to the answer of your questions:

> Just a question to this topic: When downloading CRL is done once a day by a cron job and put in the crl directory. Does rauc validate it correctly? I mean especially the usage of an indirect CRL (signed by another root ca)? And of course is it rejected (not soft fail) when a crl is expired?

Yes, in general, RAUC will check the CRL (also  see the tests that were added as part of #561) if the following conditions are met:

0) A recent RAUC version is used (but at least v1.3)
1) `check-crl` is enabled in `[keying]`section
2) `directory` is properly set in `[keyring]` section.
3) Downloaded crl files are placed under an appropriate name (see description in `directory` paragraph [here](https://rauc.readthedocs.io/en/latest/reference.html#system-configuration-file))

I am not fully sure about indirect CRLs, but as we set `X509_V_FLAG_EXTENDED_CRL_SUPPORT` it should work according to https://www.openssl.org/docs/man1.1.0/man3/X509_VERIFY_PARAM_set_depth.html.

However, you should of course test this all before shipping. To be sure that everything was set up properly and works as expected.

jluebbe, 2022-04-22:
It looks like CMS also supports including a CRL, which (when combined with a short CRL validity period) would be another way to address this use-case.

I'll close this issue for now, as there are many different ways X.509 can be used and RAUC doesn't need special handling here.