rauc/rauc #352: Installing bundle from http fails

FabianKnapp, 2018-10-22:
I tried to install a bundle from a http ressource with

`rauc install http://192.168.2.4/foobar.raucb`

`rauc-Message: installing http://192.168.2.4/foobar.raucb: installing
rauc-Message: installing http://192.168.2.4/foobar.raucb:   0% Installing
rauc-Message: installing http://192.168.2.4/foobar.raucb:   0% Determining slot states
rauc-Message: installing http://192.168.2.4/foobar.raucb:  20% Determining slot states done.
rauc-Message: installing http://192.168.2.4/foobar.raucb:  20% Checking bundle
rauc-Message: installing http://192.168.2.4/foobar.raucb:  40% Checking bundle failed.
rauc-Message: installing http://192.168.2.4/foobar.raucb: 100% Installing failed.
rauc-Message: installing http://192.168.2.4/foobar.raucb: LastError: Failed to download bundle http://192.168.2.4/foobar.raucb: Transfer failed: Failed writing body (0 != 5792)

Installing `http://192.168.2.4/foobar.raucb` failed`

/tmp/_download.raucb has always exactly 8MB size but foobar.raucb is 50MB
I think this is a libcurl specific error but dont know how to fix it. Any ideas?

Im using rauc 0.4

jluebbe, 2018-10-22:
You need to set the max-bundle-download-size in the [system] section of [system.conf](https://rauc.readthedocs.io/en/latest/reference.html?highlight=max-bundle-download-size).

I'd suggest handling the download outside of rauc, in the same custom process which controls when the update is installed. (except when using the experimental casync support).

closed 2018-10-22 by FabianKnapp

ejoerns, 2022-11-16:
@nadhem09 you might want to have a look at https://rauc.readthedocs.io/en/latest/advanced.html#http-streaming

Feel free to open further questions as [Discussions](https://github.com/rauc/rauc/discussions).