rauc/rauc #353: RFC: ignore validity time of signature in certain use-cases

mhei, 2018-10-25:
I already noticed, that the Design Checklist mentions the certificate validity periods and proposes "Validity period is large enough". However, I'd like to share my use-case and would be happy to hear other oppinions:
I want to use rauc to update embedded device via network (online) or with an USB pen drive. In the first case, we expect the embedded device to have network access and thus have a valid time by NTP or similar. 
In this case, I also can use USB update, e.g. the bundle is placed on root of USB pen drive, udev magic fires mounting and invoking rauc etc.
In case my embedded device is offline, I do not have valid time, but system time hangs in the good old 70's...
I guess there is nothing new up to this point for all people reading here.

However, I would like to allow USB updates in this case, too. I also like the signature check etc - but I'm wondering whether it would be possible to tell rauc to ignore all time related stuff at this moment, and only in this moment when installing from USB.
Ideally, this should be an command line parameter flag when calling `rauc install ...`, so other installation methods are unaffected.
What do you think about such a "feature"?

mhei, 2018-11-03:
It looks like the new `use-bundle-signing-time` could do the trick. Will test it...

ejoerns, 2018-11-06:
Well, there are at least some good guesses for setting initial time to be after the 70's (e.g. some pre-set it to be at ltest BSP build time), but that of course does not really solve your issue.

At the moment, when using D-Bus, we have not option of passing further arguments to the Install method. Otherwise we could use the '--no-verify' switch as we have for 'rauc info'. This would only work when RAUC is compiled without service support.
A later version of the D-Bus API (refer to #248) might allow this, but currently there is no project-driven development on this.

The `use-bundle-signing-time` is ATM probably the best solution to your problem (if you want to avoid erts with 'infinite' validity); with the drawback that this will be used both for the network and the USB case.

jluebbe, 2018-11-06:
> The `use-bundle-signing-time` is ATM probably the best solution to your problem (if you want to avoid erts with 'infinite' validity); with the drawback that this will be used both for the network and the USB case.

Note that this will make limited certificate validity periods basically useless, because anyone who obtains access to a key can just set the matching signing time. Alternatively set ``default_startdate`` and ``default_enddate`` for the certificate creation to the values representing unlimited validity:
```
default_startdate = 19700101000000Z
default_enddate = 99991231235959Z
```

I'd suggest fixing your system time, if possible. For example systemd will update the system time if it is earlier than the time when the systemd binary was compiled. In other cases you could check that the most recent mount timestamp is in the past (this will at least guarantee monotonic time).

mhei, 2018-12-20:
Just for the record: I finally found time to test whether `use-bundle-signing-time` works for my scenario and yes, it does (as expected).

Since I'd like to keep full validity period handling for the network case, I do not want to play with start and end date during certificate creation (ok, another approach would be to use a dedicated (modified) CA for USB case and install both CA certificates in the keyring - but this would prevent to use the same file for both installation paths - not nice either).

Fixing the system time in a standalone-offline system is difficult. I know about systemd approach, but don't see any real benefit: it only shifts the time in the right direction but makes it even harder for other components to detect whether the system time is valid or not (I mean: when kernel starts at 1970-01-01 and the embedded system will run 10 year without power outage, then the system thinks it's still in the 1980's and thus software can check for valid date with a fixed "threshold date" ... say 2018-01-01. When systemd is used, the threshold is suddenly a moving one depending on your build time... so I don't like this systemd approach very much - however, YMMV).
On the other hand one may argue that using X.509 based stuff in systems without access to a time source is broken by design.

So long story short: I would leave this issue open for now, maybe we can implement it when D-Bus API is improved.

jluebbe, 2023-10-24:
So far, we don't have any way to influence the bundle signatures via the D-Bus API and I'd prefer to keep it that way.

My suggestions in https://github.com/rauc/rauc/issues/353#issuecomment-436243609 are still the best I can offer.

closed 2023-10-24 by jluebbe