rauc/rauc #368: Asymmetric setup (recovery & normal)

vishwanathchandapur, 2018-11-20:
It is possible to get examples on Asymmetric setup.

Thanks 
Vishwa

ejoerns, 2018-11-20:
Good to hear, can we get some? :smirk: 

If I read your statement as a question instead, it would be helpful to know a bit more detailed what you expect? A simple example could be what we have in the tests here:

https://github.com/rauc/rauc/blob/master/test/install.c#L442

Or, in a minimal way:

``` ini
[slot.rescue.0]
bootname=rescue
device=/dev/null

[slot.rootfs.0]
bootname=system
device=/dev/null
```

Also note that we have a bootloader slot selection fix for asymmetric setups in the queue in #356.

jluebbe, 2018-11-20:
@ejoerns With #356 merged, we still need some real documentation on how bootnames work and are intended to be used, especially with regards to asymmetric setups.
It should probably go into https://rauc.readthedocs.io/en/latest/integration.html#interfacing-with-the-bootloader.

ejoerns, 2018-12-11:
@vishwanathchandapur, @jluebbe both an example as well as some more specific notes on `bootname` handling are now provided by #380. Please have a look.

PR #380 (2018-12-12): Slot Config Examples And More Explicit `bootname` Handling
This provides some useful examples for common slot configuration scenarios and hopefully slightly clarifies the handling of `bootname` slot config keys.

Fixes #368
closed 2018-12-12 by jluebbe