rauc/rauc #371: Is it possible to handle an ext4.gz image in a bundle?

krutzio, 2018-11-27:
Hi,

I created successfully a bundle with a packed ext4-rootfs (image.ext4.gz).

manifest.raucm: [image.rootfs] filename=image.ext4.gz
system.conf: [slot.rootfs] type=ext4

Reading https://rauc.readthedocs.io/en/latest/integration.html#slot-type, I expected that rauc determines
the file extension and knows how to handle the extraction. But running 'rauc install BUNDLE' throws the following error:
LastError: Handler error: Unsupported image /run/rauc/bundle/image.ext4.gz for slot type ext4

Can you give me a hint?

ejoerns, 2018-11-27:
Hi @krutzio,

do you really need a .ext4.gz?
RAUC support .ext4 images and they will be compressed anyway when being part of the bundle squashfs.

For ext4.gz there is no handling defined and I cannot see any benefit from supporting this.
Correct me if I'm wrong.

krutzio, 2018-11-29:
Hi, thanks for the quick reply!

There is a (little) benefit for archiving artefacts. Here is the use case:
I want to produce and update a device with the same artefact and check/compare the checksums (important for documention or license clearing) and I want to produce with an ext4 image. The bundle checksum in manifest.raucm is already there, but the bundle handles ext4 and not ext4.gz.
It would be more efficient to use an ext4.gz as artefact instead of an ext4 because it uses less HD space and a checksum test can be made before unpacking it.

So for now I have to work around:
- use tar.gz for production
- scripting around

ejoerns, 2018-12-03:
Unsure if I got exactly what you intend, but on the target the disk space usage will be the same for ext4 and ext4.gz.
Where does your size limitation during production originate from?
And why not simply also using RAUC for production / initial population of the device after partitioning?
Then you have both a valid checksum and a compressed artifact.

The use of a checksum test should also be independent on whether having an ext4 or an ext4.gz.

krutzio, 2018-12-06:
I am not talking about the size on target (which is of course limited), I am talking about the size of an archived artefact (production image or archive) on a server. So there is no size limitation at all, only a better disk usage on the server.

Using RAUC for production would simplifiy the production workflow, but not solve the use case. Let me resume it:
- produce and update with the same artefact (can be an image or an archive), so it has the same checksum
- produce with an ext4 (so the artefact can only be ext4 or ext4.gz)
- i want to save space on server HD, where artefacts are archived

Is it clear now?

I know how to handle the workarounds, so if you don't see a benefit, please close the question.

ejoerns, 2018-12-10:
@krutzio you could archive the bundle instead but I'm not sure if that fits your needs.

Or you use a compressed tar archive, then you have it compressed both inside the bundle and where you store your artifacts.

Third option would be to use casync, then you only chunks of content being different to your other releases needs to be stored additionally.

For now I do not see this mainline because it adds more overhead than a real benefit. You would have to pipe the output of the ext4 unpacking directly into the target slot to handle this properly.

Maybe a custom install hook could handle this, too.

krutzio, 2018-12-11:
@ejoerns ok, thanks.

closed 2018-12-11 by krutzio