rauc/rauc #374: Marking slot active / good fails.

FabianKnapp, 2018-12-04:
`rauc status mark-active` fails with:

`rauc-Message: rauc mark: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30: failed to activate slot rootfs.0: Failed marking 'rootfs.0' as primary: Failed to start barebox-state: Too many open files`

However, `barebox-state -s bootstate.system0.priority=10` works and `rauc status` returns now good for system0.

Maybe it is only possible to call `rauc status mark-active` one time?

I have a helper service, that steadily checks if the system health is good and change the RAUC status of the current slot accordingly (if needed).
The problem persists after stopping this service.

Any ideas?

FabianKnapp, 2018-12-04:
Maybe this is fixed with commit 7665b1559876493bafb48246c4c33f4011bfbcf2 ?

ejoerns, 2018-12-05:
Indeed, this sound like it could be a result of what was fixed in https://github.com/rauc/rauc/commit/7665b1559876493bafb48246c4c33f4011bfbcf2.

I do not see any point in setting `LimitNOFILE=` as this should not be required in case this was a bug, and otherwise I cannot see any valid value for this limit that would be more appropriate than defaults.

FabianKnapp, 2018-12-05:
Unfortunately, I cant reproduce this error. I will patch 0.4 temporarily with 7665b15 and use `LimitNOFILE=8192` in rauc.service for safety.

...looking forward for 1.0 release :1st_place_medal:

ejoerns, 2018-12-10:
Ok, and we will try to push 1.0 to come out soon... ;)

I'll close this for now.

closed 2018-12-10 by ejoerns