rauc/rauc #375: version handling in git snapshot improvable

ukleinek, 2018-12-10:
Hello,

if rauc is build e.g. from https://github.com/rauc/rauc/archive/v1.0-rc1-59-g0745c7bf6c78.tar.gz there is no version input for the `git-version-gen` script yielding

    root@$machine:~ rauc --version
    rauc UNKNOWN

Something like `v1.0-rc1` or `v1.0-rc1+` even would be great here.

jluebbe, 2018-12-10:
The tar archives automatically generated by github do not contain the necessary information. Either build from a git checkout or use a (pre-)release tar archive.

closed 2018-12-10 by jluebbe

ukleinek, 2018-12-13:
I just noticed that --fallback doesn't work as intended. The reason is the following in `git-version-gen`:

    [...]if test "x$fallback" = x || git --version >/dev/null 2>&1; then
        v=UNKNOWN
    else
        v=$fallback
    fi

So the fallback is never used if git is properly installed.