rauc/rauc #376: Create Image-Based Delta Bundles

ejoerns, 2018-12-10:
For setups where one has both a root-filesystem slot (rootfs) and an application slot (appfs), one might want to update the appfs much more frequent than the rootfs.

While the aim of RAUC is to install only well-defined sets of software by providing all slots of a slot group, one would currently need to provide images for both appfs and rootfs in each bundle.
On the other hand, there is the requirement to not let bundles become unnecessarily large and being able to omit the rootfs slot.

A proper compromise for this can be to implement image-based delta bundles or 'dummy images'.

The basic concept is to still reference the exact (tested) state of the rootfs with an bundle containing an appfs-only update, but not providing an actual rootfs image in the bundle.

Then RAUC will simply skip the installation of the rootfs on slots that already have the right rootfs version and reject installation on those that have a wrong rootfs version.

This combines the benefit of being explicit across the entire system state and having small bundles by omitting non-required images.

gajanak, 2019-04-13:
Perhaps for some upgrades my feature fork will help.
It allows extract a tar to an already mounted partition.
For example, if you want only add or replace some files.
But this will fail in some situation... where previous state of slot is not what expected. ;)
You can look at (https://github.com/gajanak/rauc).
I can imagine to add an option to manifest giving an slot hash, to restrict the upgrade ?

tte-paul, 2021-04-23:
Hi Enrico, just wondering how is the image-based delta bundles going? @ejoerns

ejoerns, 2021-04-23:
Hi @tte-paul. Huh, haven't been working on it for too long as it was dropped from the requirements list of the project where I originally started creating it for. But always wanted to finish it.
Do you have a use case for it?

tte-paul, 2021-04-25:
Hi @ejoerns, our use case is to split the main application from the root file-system which means we will have two slots (rootfs + appfs) and most of the time we only need to update appfs. Due to the limit of network bandwidth we want to put only appfs into the bundle for downloading. 

This case is just as what you described in the beginning of this issue. What would be your suggestion?

Sourabh1107, 2021-11-02:
Hello @ejoerns, can you please let me know if the above approach can be implemented or there any drawbacks to it?

jluebbe, 2021-11-15:
Discussion https://github.com/rauc/rauc/discussions/794 is related to this.

ejoerns, 2023-02-21:
> The basic concept is to still reference the exact (tested) state of the rootfs with an bundle containing an appfs-only update, but not providing an actual rootfs image in the bundle.

> Then RAUC will simply skip the installation of the rootfs on slots that already have the right rootfs version and reject installation on those that have a wrong rootfs version.

With RAUC's built-in HTTP(S) streaming support, image-based delta bundles move out of focus since when using slot skipping, the skipped image will not be downloaded at all.

commit 7319c39b (2023-02-21): docs/basic: add note on combining slot skipping with bundle streaming

Closes #376