rauc/rauc #382: RAUC service permanent CPU load ~50% if pre-install handler is blocking.

FabianKnapp, 2018-12-12:
**Affects version:** 0.4

**Description:**
A pre-install handler script that is blocking (sure, this should not happen) leaves the RAUC service in constant high CPU usage until `systemctl restart rauc` is invoked.

**Solution Idea:**
Maybe it would be good to have a configurable timeout for handler scripts. Since the update event never returns if a handler is blocking with the current implementation.

jluebbe, 2018-12-12:
Could you attach gdb to the running rauc process in that situation and get a backtrace? It should be blocked in g_subprocess_wait_check() and not use any CPU time while the hook is still running.

ejoerns, 2018-12-12:
What actually happens in this case is:

```
# mkfs.ubifs: UBI volume is not empty.  Format anyways? (y/N) failed to read prompt; assuming 'no'
# unknown response; please try again
# mkfs.ubifs: UBI volume is not empty.  Format anyways? (y/N) failed to read prompt; assuming 'no'
# unknown response; please try again
# mkfs.ubifs: UBI volume is not empty.  Format anyways? (y/N) failed to read prompt; assuming 'no'
# unknown response; please try again
# mkfs.ubifs: UBI volume is not empty.  Format anyways? (y/N) failed to read prompt; assuming 'no'
# unknown response; please try again
```

And CPU usage steadily increases...

FabianKnapp, 2018-12-13:
I would prefer to abort the pre-install handler if an input is required or at least introduce a timeout since the update would stuck infinite.

xyzzy42, 2018-12-19:
What happens if using `mkfs.ubifs /dev/data-inactive <&-` instead?  That will close stdin instead of redirecting /dev/null to it.  Perhaps what will cause ubifs to abort instead of retrying to read from /dev/null?

ejoerns, 2018-12-20:
@FabianKnapp I have no idea how to detect that a program waits for input in this case, but anyway, all other mkfs variants apparently handle this case properly and switch to non-interactive mode when no input can be read. Maybe something one should fix in mkfs.ubifs.

Concerning a timeout, yes, it would be better if RAUC would not completely stall in case of calling a infinitly blocking process, but then comes the question about the valid period of such a timeout.
Things that could be done are quite free and operations, especially on embedded systems, can be quite long-running.
I don't say this is not worth tinking about it, but I am not sure of what default timeout one should use.

@xyzzy42 the shell call above was more an example to reproduce it, the actuall mkfs.ubifs invocation is performed by a g_subprocess_launcher in rauc and must be valid for all cases.

FabianKnapp, 2019-02-28:
> @FabianKnapp 
> Concerning a timeout, yes, it would be better if RAUC would not completely stall in case of calling a infinitly blocking process, but then comes the question about the valid period of such a timeout.
> Things that could be done are quite free and operations, especially on embedded systems, can be quite long-running.
> I don't say this is not worth tinking about it, but I am not sure of what default timeout one should use.
> 

Maybe it is worth thinking about a timeout variable in system.conf for handler scripts that defaults to infinite if not explicitly set. However, I think this is more an additional feature since it is a more a problem with mkfs.ubifs.

closed 2023-10-24 by jluebbe